openpnp/openpnp #617: CONNECT_COMMAND send to main driver on every connect, to subdriver only on first connection after startup

mgroel, 2017-08-04:
# Problem Report
## Describe the Issue
Noticed, that the connect command will not be sent to a gcode-subdriver on second and more connections if they are made in the same runtime of openpnp. To the main gcode-driver the connect_command is sent every time as expected.

## Steps to Reproduce
1. Start OpenPnP, watch the communication in DEBUG/TRACE
2. Click the Power On button. -> connect-commands are sent to both drivers.
3. Click the Power Off button -> disable-commands are sent to both drivers.
4. Click the Power On button again -> the connect commands are sent only to main driver.
5. Close OpenPnp, start at step 1

## Expected Result
Both drivers should sent off the connect-command, don't they?

## Actual Result
Only main-driver gets connect-command every time. Subdriver only on first start after opening OpenPnP

vonnieda, 2017-08-04:
Can you attach your machine.xml and log file please?

mgroel, 2017-08-04:
[log.txt](https://github.com/openpnp/openpnp/files/1200591/log.txt)

[machine.xml.txt](https://github.com/openpnp/openpnp/files/1200592/machine.xml.txt)

vonnieda, 2017-08-04:
Thanks @mgroel. I'm not seeing anywhere in that file where the connect command is being sent again. From setEnabled(true) (power button on), the CONNECT_COMMAND and the ENABLE_COMMAND for each driver is sent to each driver, then there is a setEnabled(false) (power button off) and DISABLE_COMMAND is sent, then a final setEnabled(true) (power button on) where only ENABLE_COMMAND is sent.

This all looks correct to me. The expected behavior is that CONNECT_COMMAND is only sent when connecting to the controller the first time. This happens during the first "power on" after starting OpenPnP. OpenPnP does not disconnect from the controller during "power off", so only the ENABLE/DISABLE commands are sent for further power on / power off events.

mgroel, 2017-08-04:
Ahhhh, sorry for that.
I wondered what the actual difference between connect and enable was. Didn't notice, that the connection is not released until closing openpnp. Now everything is clear. sorry for bothering you.

Closed :)

closed 2017-08-04 by mgroel