openpnp/openpnp #618: Problem Report. Camera Location Button.

terryjen, 2017-08-04:
# Problem Report
## Describe the Issue
After set the position of the bottom Camera(Coordinates:240, 90,10, data is not accurate), and I home the PnP (Coordinates 400,300,0), and press the button in the Machine Setup->Cameras->Bottom Camera->Camera Specific-> Location-> "Position the tool over the center of the location", it only moves a very small step, and reset the location to similar to the coordinates where bottom Camera is, while the actual position is almost same as to Home Position.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Home the Machine.
4. Move the nozzle to the bottom Camera position. 
5. Set the Location of the Camera Positon in Camera Specific.
6. Home Again.
7. Press the button "Position the tool over the center of the location" in the Camera Specific->Location.

## Expected Result
The nozzle shall move to the bottom Camera Position.

## Actual Result
It only moves very little, near the Home, and timeout error.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Here is the log file:
2017-08-04 11:53:13 ReferenceHead DEBUG: H1.moveToSafeZ(1.0)
2017-08-04 11:53:13 ReferenceNozzle DEBUG: N1.moveToSafeZ(1.0)
2017-08-04 11:53:13 GcodeDriver DEBUG: sendCommand(G0   Z0.0000  F1000 ; Send standard Gcode move, 10000)...
2017-08-04 11:53:13 GcodeDriver TRACE: [COM3] >> G0   Z0.0000  F1000 ; Send standard Gcode move
2017-08-04 11:53:13 GcodeDriver TRACE: [COM3] << ok
2017-08-04 11:53:13 GcodeDriver DEBUG: sendCommand(COM3 G0   Z0.0000  F1000 ; Send standard Gcode move, 10000) => [ok]
2017-08-04 11:53:13 GcodeDriver DEBUG: sendCommand(M400 ; Wait for moves to complete before returning, 10000)...
2017-08-04 11:53:13 GcodeDriver TRACE: [COM3] >> M400 ; Wait for moves to complete before returning
2017-08-04 11:53:13 GcodeDriver TRACE: [COM3] << Limit switch min_z was hit - reset or M999 required
2017-08-04 11:53:13 GcodeDriver TRACE: [COM3] << ok
2017-08-04 11:53:13 GcodeDriver DEBUG: sendCommand(COM3 M400 ; Wait for moves to complete before returning, 10000) => [Limit switch min_z was hit - reset or M999 required, ok]
2017-08-04 11:53:13 ReferenceCamera DEBUG: TopView-Head.moveToSafeZ(1.0)
2017-08-04 11:53:13 ReferenceActuator DEBUG: VacSolenoid1.moveToSafeZ(1.0)
2017-08-04 11:53:13 ReferenceActuator DEBUG: ExhSolenoid2.moveToSafeZ(1.0)
2017-08-04 11:53:13 ReferencePasteDispenser DEBUG: D1.moveToSafeZ(1.0)
2017-08-04 11:53:13 ReferenceNozzle DEBUG: N1.moveTo((240.870000, 90.960000, 0.000000, 0.000000 mm), 1.0)
2017-08-04 11:53:13 GcodeDriver DEBUG: sendCommand(G0 X240.8700 Y90.9600   F1000 ; Send standard Gcode move, 10000)...
2017-08-04 11:53:13 GcodeDriver TRACE: [COM3] >> G0 X240.8700 Y90.9600   F1000 ; Send standard Gcode move
2017-08-04 11:53:13 GcodeDriver TRACE: [COM3] << !!
2017-08-04 11:53:23 MessageBoxes DEBUG: Error: java.lang.Exception: Timeout waiting for response to G0 X240.8700 Y90.9600   F1000 ; Send standard Gcode move
2017-08-04 11:57:08 SystemLogger INFO: Don't call wizardCompleted

Maybe some of my understanding of the calibration is wrong, hope you can help me.
Thanks.
Terry

vonnieda, 2017-08-04:
You appear to have a smoothie setup issue. Look at the logs - it is
reporting that your z limit switch is hit, which means the controller must
be rehomed or reset.

terryjen, 2017-08-04:
Yes, it hit the switch and stopped, but the coordition on the status bar changed to the bottom camera position. 
And I'm wondering why this happen, when Home, the coordinate for example is (400,300,1), and the bottom camera position is (245, 100, 50) for example, click that button shall move to camera postion. but why will it goes to the oppsite direction and hit the Z limit switch.
When I'm at Home, I send the command "G0 X245, Y100, Z50". it is fine.

vonnieda, 2017-08-04:
Please attach your machine.xml and the complete log file, including the homing operation.

terryjen, 2017-08-04:
Sorry, I already left the lab, I will upload these files after long weekend on Tuesday.
Thanks, Mr. Jason von Nieda

terryjen, 2017-08-08:
Hi Mr. Jason von Nieda,

Thanks for your time to take a look at the following pictures and files.

**we can get the home coordinates:**
[image]

**when press the button: "Position the tool over the center of the location",** 
[image]

**This is the log file, only two steps:**
[log.txt](https://github.com/openpnp/openpnp/files/1208172/log.txt)

**This is the machine.xml**
[machine.xml.txt](https://github.com/openpnp/openpnp/files/1208173/machine.xml.txt)

Thanks very much!
Terry

vonnieda, 2017-08-08:
It looks like you may have your Z axis inverted. See https://github.com/openpnp/openpnp/wiki/User-Manual#coordinate-system. Z should move negative down, positive up. 

The root of the issue is that OpenPnP is attempting to move Z to SafeZ, which is 0. Since your limit switch at somewhere between 1 and 0 it's being hit during this operation.

You can change SafeZ in Machine Setup by selecting each head mounted object: nozzle, camera, actuator, etc.

With that said, it would be better to invert your Z axis logic so that negative is down and positive is up, and have 0 be fully up. Then all of your Z coordinates will be negative and you should not see this error.

If you have further questions about this, please post to the mailing list at http://groups.google.com/group/openpnp. The issues section is only for bugs or feature requests, and this is just a configuration error.

closed 2017-08-08 by vonnieda