openpnp/openpnp #621: Warning about Parts outside of the Board

KarlZeilhofer, 2017-08-16:
# Feature Request
## Describe the Feature
Give a warning about parts outside the PCB

## How Will It Look?
### When starting a job:
Popup message with  
Warning: "Some parts are outside of the boundaries of the board. Do you want to continue?" 
Buttons: Yes, No
### When importing a placement file
Warning: "Some parts are outside of the boundaries of the Board"
Button: OK
This could be suppressed, if the boards dimension isn't set (widht=0 and length=0)

## Why Do We Need It?
1. What problem does it solve?
We had the case, that there was a problem with exporting the .pos file from KiCad. The auxilary origin wasn't set correctly, and therefore all parts were outside the boundaries of the given board dimensions. 

During setup, we wanted the top camera to go above a part, and then the machine drove against an end switch. 

2. Is it useful for everyone, or does it only solve a problem for your machine?
I think, it is for everyone.

cri-s, 2017-08-19:
A lot of users use negative centeroid or where the board zero is on board center.
It works tweaking the board size, as example if having negative centroids, board width must be 0 and
if using board center zero, width must be set to half of the real board size.
Further there are users working with negative centeroid data and negative global coordinates, having 
0 on the back right edge of pnp machine instead of the front left edge.
Further i have discovered that a lot of users don't use board height/width at all.

Personally i think, softLimits make more sense to handle this case.

KarlZeilhofer, 2017-08-20:
BTW, is there anywhere a setting about the size of the machine?

I drive our head from time to time accidentially into the endswitches, whitch makes a comlete restart of openpnp necessary...

vonnieda, 2017-08-20:
@KarlZeilhofer Not currently, but the issue has been raised: https://github.com/openpnp/openpnp/issues/308