openpnp/openpnp #623: Crash when perform a feed or a feed and pick operation

johngrabner, 2017-08-19:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Slowly working on converting my TVM802 to OpenPnP.
Have gotten to the stage of JOG controls work and defined a reference drag feeder.
Move camera to feeder and move tool to feeder functions as expected.
The feed operation causes a crash,
https://gist.github.com/anonymous/e86ef0b702e488464fb6ace726e5b3b6

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Home
4. in feeder tab, click feed icon -> crash

## Expected Result
A feed operation

## Actual Result
a crash.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
https://gist.github.com/anonymous/e86ef0b702e488464fb6ace726e5b3b6

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

johngrabner, 2017-08-19:
version of OpenPnP is # 2017-08-16

vonnieda, 2017-08-19:
By crash do you mean the software window closes and you have to restart, or
are you talking about a dialog box with an error and you can continue?

johngrabner, 2017-08-19:
Looks like update 2017-08-16 was a breaking change. Deleted the drag feeder and recreated, and now it no longer crashes.

johngrabner, 2017-08-19:
The crash was a java traceback diaglog. Image of the traceback was in https://gist.github.com/anonymous/e86ef0b702e488464fb6ace726e5b3b6

vonnieda, 2017-08-19:
Hi @johngrabner, what I am trying to discern is whether the software exited and had to be restarted, or if it was just a dialog that you were able to close and move on. It makes a difference in how I approach the problem. 

If it's just a dialog with an exception stack trace, that's not a crash, it's just a captured exception which would typically indicate misconfiguration. Whereas if the software actually closed and had to be restarted, that's a crash and may indicate a programming error.

I don't see a stack trace in the log you posted, but I do see a NullPointerException related to template matching. I think I probably know what went wrong. Did it look like this?

[image]

richard-sim, 2017-08-19:
I expect it was due to having an empty rect for the vision of that feeder, essentially making it blind:
               <template-image-top-left units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
               <template-image-bottom-right units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>

vonnieda, 2017-08-19:
Yep, that's how I reproduced it. I need to add a check for when use vision
is on but there is no template image.

Jason

commit 18ef6972 (2017-08-19): Fix #623 by providing better error messages when feeder is misconfigured.
closed 2017-08-19 by vonnieda