openpnp/openpnp #624: Add property "align to long/short axis" to CV stage MinAreaRect

KarlZeilhofer, 2017-08-21:
# Problem Report
The result of the MinAreaRect is not predictable in terms of the rotation. Sometimes north is in the direction of the long side, sometimes it is in the direction of the short side. If the current results are used for alignment, the 90° angle is wrong with a chance of 50%. 

## Steps to Reproduce
1. Put a simple resisotor or capacitor onto the nozzle
2. check alignment with the bottom vision and draw the orientation of the detected rect
3. repeat 2. for different nozzle angles. 

## Expected Result
I would expect, that the orientation is always along the main direction (long side) with an option, to align with the short axis. 

## Actual Result
The 90° angle is not predictable. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

[image]

[image]

[image]

KarlZeilhofer, 2017-08-21:
here the pipeline, i'm working with:
```
<cv-pipeline>
   <stages>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageCapture" name="0" enabled="true" settle-first="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="13" enabled="true" prefix="bv_source_" suffix=".png"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.BlurGaussian" name="10" enabled="true" kernel-size="9"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="4" enabled="true" diameter="100"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="1" enabled="true" conversion="Rgb2Gray"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.Threshold" name="12" enabled="true" threshold="120" auto="false" invert="false"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.FindContours" name="5" enabled="true" retrieval-mode="List" approximation-method="None"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.FilterContours" name="9" enabled="true" contours-stage-name="5" min-area="500.0" max-area="1000.0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="11" enabled="true" diameter="0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawContours" name="7" enabled="true" contours-stage-name="9" thickness="2" index="-1">
         <color r="255" g="255" b="255" a="255"/>
      </cv-stage>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MinAreaRect" name="result" enabled="true" threshold-min="100" threshold-max="255"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRecall" name="14" enabled="true" image-stage-name="0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawRotatedRects" name="8" enabled="true" rotated-rects-stage-name="result" thickness="2" draw-rect-center="false" rect-center-radius="20" show-orientation="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="15" enabled="true" prefix="bv_result_" suffix=".png"/>
   </stages>
</cv-pipeline>
```

cri-s, 2017-08-21:
What is the reason of this issue ?
ShowOrientation inside drawRotatedRects could be useful for debuging inside looseFeeder, 
and that is the reasone why this parameter exist. It is a debugging parameter and it could be used
outside debug where it make sense, here it don't make sense.

vonnieda, 2017-08-21:
This is true, but it is handled correctly inside the bottom vision code. It
assumes a part is never picked more than 45 degrees from correct, so
normalizes the angle with that in mind. That being said, I would support
adding a Boolean to this class that would allow you to normalize the
rectangle as long as it was optional and defaulted to false. I think we
also have a stage that normalizes rectangles already, too, but I can't
check at the moment.

Jason

cri-s, 2017-08-21:
As vision gives back two valid angles, it cannot determine if component at 90 or 270 degree,
it always jump from one to the opposite side after normalisation.

KarlZeilhofer, 2017-08-22:
The problem existed with the ReferenceLoosePartFeeder. It picked up the parts correctly, but the orientation of the placement was randomly correct and by 90° wrong. So I had to use the AdvancedLoosePartFeeder, which works really good, but it is much more effort to get it up and running. 

A simple prealignment to the main axis of a rectangle costs de facto no processing time, and solves some problems. Why should an algorithm return a random number, when it could easily relate it to a main feature of an object?

Thank you Jason, for the pointing out, that the bottom vision code corrects for ±45, I didn't know that. 
@cri-s I'm not sure, of which situation you are talking, but for a rectangle it should be valid to limit the angle to 0-180°.

cri-s, 2017-08-22:
Bottom vision correct for 45 deg. , this is +-22.5 deg.
That you don't understand the angle value retuned from vision pipeline is
ok and not a problem. You don't need understand that.

As bottom vision correct 45 deg  looseFeeder correct 90 deg.  Fill a Bug
report that it need correct 180 deg and probably Jason fix it.

cri-s, 2017-08-22:
Inside LooseFeeder.java

Instead of:
 // Get the result's Location
        // Update the location with the result's rotation
        location = location.derive(null, null, null, result.angle);

this should work:
 // Get the result's Location
        // Update the location with the result's rotation
        location = location.derive(null, null, null, (result.size.width > result.size.height)?result.angle+90.:result.angle);

Cri

vonnieda, 2017-10-26:
This is now fixed via #666.

closed 2017-10-26 by vonnieda