openpnp/openpnp #625: Broken Serial Connection after Restart of Motion Controller

KarlZeilhofer, 2017-08-22:
# Problem Report
## Describe the Issue
When connection to a USB-serial device is lost (for example the head drove into an end-switch, and the TinyG had to be restarted), there seems to be no chance, to continue the current job. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Turn off the USB serial device (e.g. TinyG)
4. Turn it back on
5. Now the communication is broken
6. Try to disable the machine via the button below the home button (this works, if no commands are set in the G-Code driver). This step should also close the serial device within OpenPnP. 
7. Change the serial device in the driver setup to the current one (it was /dev/ttyUSB0, after repowering TinyG it is /dev/ttyUSB1, since OpenPnP still had the device opened)
8. Try to enable the machine again - no connection possible. 
9. It seems, OpenPnP doesn't open the new configured device, but sticks with the old serial device (/dev/ttyUSB0)

## Expected Result
On Enable:
OpenPnP should close any open serial devices, load current settings from the driver open the new serial device. 

## Actual Result
The communication is broken. The new device is not opend after reconfiguration of the serial device in the gcode driver.

vonnieda, 2017-08-22:
There is a reason for this behavior, but I think we can make it an option. The reason is that many Arduino based controllers do a full reset upon connect, losing all their current status. When I was first developing the serial based drivers I thought it best to be able to disable/enable without reseting every time.

But, I agree. This is a problem now for the majority of users. I think we should add a checkbox ""Reconnect on Enable?" and do this functionality.

cri-s, 2017-08-22:
You must write a udev rule to make the USB controller permanently use the same /dev/ttySmoothie and then make softlink as example from ttyS33 to ttySmoothie in order it is recognized as valid serial name if that is not the case for ttySmoothie.
There are plenty of instructions how to do on WWW and similar thing exists on Windows. This way it retain always the same com device and can always reconnect. Reconnecting automatically as Vonnieda suggest is s security risk but possible using libusb and scanning for video/paid and serial number. Probably it destroy nozzles sometimes if user forget to disable autosuspend. Mac probably follow Unix example.

vonnieda, 2017-08-22:
Why would the serial port name change between OpenPnP enable/disable?

cri-s, 2017-08-22:
Because it change. On win, and Linux, don't know the mac behavior.

cri-s, 2017-08-22:
It change if USB get unconnected and reconnect , the reconnect could have
several reasons.

vonnieda, 2017-08-22:
I have never seen a case where it changes unless the device is unplugged
and replugged. It definitely doesn't happen on mac and I'm pretty sure
about windows. This request does not refer to unplugging the controller -
simply enable/disable in OpenPnP.

vonnieda, 2017-08-22:
If what you are describing was true a user would have to reselect the
serial port every time they start OpenPnP, and that is not happening.

cri-s, 2017-08-22:
That is the behavior I have seen often. Java have device open as it should, USB make new initial request, and receive next device number as the previous one is looked/open by java program as there is no event that checks the USB disconnect and free the resource inside java serial code.

cri-s, 2017-08-22:
It happens only if there are USB common problems. Interferences, USB auto sleep, overpower detected, .... .
After timeout, tens of seconds to minutes the resource get available and next time it works with old I'd.

cri-s, 2017-08-22:
Using udev rules the single device get the same name every reconnect. On windows it is possible to look serial to one USB vid/paid/serial atchi ing the same thing.

vonnieda, 2017-08-22:
Okay, then why are you saying it is a problem here? We are not taking about
USB problems, we are taking about simply serial software
disconnect/reconnect.

The problems you described would affect OpenPnP in any case - it makes no
difference if we add the ability to reconnect.

Claiming this is a security risk is ridiculous. It is not a security risk
to attempt to reconnect to the port you were just using. This is done by
nearly very program that uses serial ports all the time.

cri-s, 2017-08-22:
The most probable reason for this issue is emi problems, limit switch cause some ground/reset trouble where controller resets or simply USB need reconnect and the result is device change.  Maybe simple ferrite could help over USB cable.

cri-s, 2017-08-22:
What I claim security (injury) risk is reconnect automatically to controller not initiated by user if serial connection was broken and optionally automatically homing or not as motion controller is then in unknown state and even legacy moves like homing could hit user at higher speed.
Further it is undefined if controller have reset, firmware bug or have some valid char inside its buffer derived from bad USB communication and the next enter could execute it.

cri-s, 2017-08-22:
I read the problem above as timeout issue 
when device name changes port name after USB reconnect.

vonnieda, 2017-08-22:
> What I claim security (injury) risk is reconnect automatically to controller not initiated by user if serial connection was broken and optionally automatically homing or not as motion controller is [...]

Nobody said anything about automatically reconnecting or automatically homing or doing anything not initiated by the user.

The request is simple: disconnect on disable, connect on enable. This differs from the current behavior in that currently we only connect on enable the first time, and only disconnect on exit.

KarlZeilhofer, 2017-08-23:
Cri, thanks for your hint of a fixed name with an udev rule, but for now, this isn't the topic. I don't know if this is an EMI problem, I have to investigate this in the TinyG docs. 

The major problem is, that I lost the current Job, which is not funny on a pcb with 400+ part on it. I think it would be another GitHub issue to save and restore the current state of a job.

I don't understand, why reopening the serial device should only be optional. The user wouldn't notice this.

vonnieda, 2017-08-23:
@KarlZeilhofer The reason for making it optional is to support the comment I made in my first response: "The reason is that many Arduino based controllers do a full reset upon connect, losing all their current status."

That being said, I'm not sure that is even a very good reason anymore. Back then I was re-enabling so many times during development that it became annoying to have to re-home every time. But now, I don't think anyone would want to enable/disable and then continue without homing, so maybe the point is moot.

So, in short, I'd be fine with changing this to disconnect on disable, reconnect on enable. If users complain we can add the option later.

As for save/restore current job, that is being worked on: https://github.com/openpnp/openpnp/issues/280

cri-s, 2017-08-23:
I was not aware that gdriver don't allow it.
Using gbrl with DTR jumper disabled it is the main e-stop. It is configured
to throw error if movement without homing. Yes, its really useful.
If serial comm. disappear or power loss there is no way to automatically
recovery without editing job manually as software don't know wat was
executed and was not.

KarlZeilhofer, 2017-08-24:
I've impemented this feature now, but I have problems with the GUI, please see the commit message:

https://github.com/KarlZeilhofer/openpnp/commit/10b66179fc0cadde67dccb759042fe55f3d3e49a

It is not a major problem, but the user expects the GUI to show a disabled state, after clicking the disable button.

vonnieda, 2017-08-28:
@KarlZeilhofer I've just returned from vacation and will take a few days to catch up on everything. I should be able to get a chance to look at this before the end of this week.

vonnieda, 2017-09-03:
Hi @KarlZeilhofer, in case you didn't see it, I commented on the commit with a request for a bit of explanation: https://github.com/KarlZeilhofer/openpnp/commit/10b66179fc0cadde67dccb759042fe55f3d3e49a#commitcomment-23958970

romain145, 2019-01-23:
I confirm that Keep Alive is off.
The disable command is as follow:
[image]

Steps to reproduce:
1. Reset the controller (by sending "reset" or pressing the reset button)
2. See the USB disconnect & reconnect
3. Press the Disable button and receive the error message:

[image]

vonnieda, 2019-01-23:
@romain145 Ah, I see, sorry, I misunderstood the problem. 

So, the issue is that the disconnect never gets reached because of the timeout exception. I think we can approach this two ways:

1. Wrap the disable command and continue with the rest of the disconnect in the case of an error.
2. See if we get an event from the serial library when the port goes away, in which case we'd automatically disconnect.

markmaker, 2021-04-25:
I believe I fixed this in Advanced Motion Control and follow-up. Exceptions are now ignored during tear-down. Multi-driver setups should now also work correctly, i.e. when one connect out of many fails, the already opened connections are closed again. The machine connected state should always remain clean even across many drivers/controllers, if keep-alive is disabled, which in turn is recommended by Issues & Solutions. I tested this a lot with my test rig of three different controllers and through firmware development hiccups, flow-control troubleshooting etc. i.e. I'm quite confident it is now stable.

closed 2021-04-25 by vonnieda

heeplr, 2024-03-25:
@vonnieda 

I just encountered a similar problem where I have to restart OpenPNP after disconnect since it seems to keep the lock so that ```/dev/ttyACM0``` (in my case) remains stale and the actual device becomes ```/dev/ttyACM1``` after reconnect. (Also the machine stays enabled in the GUI.)

>     2. See if we get an event from the serial library when the port goes away, in which case we'd automatically disconnect.

This would probably solve a lot of issues related to reconnects and would also make detection of disconnects faster (without timeouting first). 
I think [LISTENING_EVENT_PORT_DISCONNECTED](http://fazecast.github.io/jSerialComm/javadoc/com/fazecast/jSerialComm/SerialPort.html#LISTENING_EVENT_PORT_DISCONNECTED) could be used for that, but I'm not sure where and how to implement it cleanly.
I believe freeing resources of the (physically disconnected) port when handling this event would release ```/dev/ttyACM0``` so it could be removed by the OS. Then after reconnect, it would be enumerated as ```/dev/ttyACM0``` again.

vonnieda, 2024-03-25:
@heeplr I'm a little out of the game at the moment, but I think there is a setting on GcodeDriver communications for "Keep Alive?" or something like that. Toggling that should fix what you are seeing, I think.

heeplr, 2024-03-27:
@vonnieda Thanks for your answer. 

Just wanted to add, that "Keep Alive" doesn't seem to fix it.
I think there's no way around than immediately freeing resources upon disconnect.