openpnp/openpnp #627: Feature: QiHe TVM802 Equivalance

johngrabner, 2017-08-24:
# Feature Request
## Describe the Feature
This ticket will inventory missing capabilities in OpenPnP to make OpenPnP a proper superset of functionality available in V2.35 of QiHe software for the TVM802. 

In many ways, OpenPnP is better than the QiHe software (my opinion). Excellent documentation, Customizable vision pipeline,  clean GUI for package and job processing are a few, and it's open.  Missing functionality, however, makes OpenPnP not a clear replacement for QiHe software.  Correcting these shortcomings will make OpenPnP a clear winner.

The shortcomings that stand out are:

1. Drag feeder with peel stepper
2. Pin down and pin down stuck detection with giggle recovery
3. Jog prevention if pin down
4. Start / Stop / Pause front panel button control
5. Status LED on front panel control
6. Jog prevention if home not done

## How Will It Look?
### Drag feeder with peel stepper

Near done with implementation on my clone copy of OpenPnP, will solicit feedback to make sure fits with rest of functionality before merging back.

It will be a superset of current ReferenceDragFeeder with extra optional parameters.

### Pin down and pin down stuck detection with giggle recovery

Near done with implementation, will solicit feedback to make sure fits with rest of functionality.

It will be a superset of current ReferenceDragFeeder with extra optional parameters.

### Jog prevention if pin down

Looking for suggestions.  It would be nice to have specialized actuators that present only relevant commands in the machine setup vs presenting all commands.  Then a specialized actuator for DragPin could be implemented and have a specific configuration parameter of "allow head movement with pin down in only these safe areas.

### Start / Stop / Pause front panel button control

Pushing the button on the front panel is convenient when processing many pcbs.
I found myself using these all the time with the QiHe software even though a button click was available on the QiHe software.

### Status LED on front panel control

Feel good LEDs. On TVM802, they are very small.  On larger scale PnP machines, a big status light is essential to direct users to machines needing attentions.

### Jog prevention if home not done

Understand that some machines are auto homed.  For those that are not, it would be extremely nice to have a parameter to present any head movement until the machine is homed.  Head movement without being homed ends up with head crashing and other bad things. Even damaging parts.

Damaging the machine is not an option if OpenPnP is to be considered as a replacement for vendor SW on commercial machines.

## Why Do We Need It?

### Drag feeder with peel stepper
What problem it solves: Manually stripping component tapes is not practical for commercial production, even small production runs.

Specific to this machine: All commercial machines have the ability to peel tapes automatically.  The TVM802 approach is a good approach for very low-cost HW.

### Pin down and pin down stuck detection with giggle recovery

What problem it solves: Head movement with pin down will damage machines.

Specific to this machine:  While a drag pin is a poor man's version of component feeders, it is essential for adoption of OpenPnP by low-cost HW suppliers. The TVM802 is sold under a couple of bands.

### Jog prevention if pin down

What problem it solves: Head movement with pin down will damage machines.
Specific to this machine:  Same as above.

### Start / Stop / Pause front panel button control

What problem it solves:  Many low-cost PnP machines have this type of control.
Specific to this machine:  Same as above.

### Status LED on front panel control
What problem it solves:  Many low-cost PnP machines have this type of control.
Specific to this machine:  Same as above.

### Jog prevention if home not done
What problem it solves:  Many low-cost PnP machines have this type of control.
Specific to this machine:  Same as above.

## Provide Examples
V2.35 version of QiHe software.

johngrabner, 2017-08-24:
Jason, you could assign this to me.  I will implement all of this in my clone, but my preference is to make this more generic and offer to OpenPnP so I do not need to re-merge continualy as OpenPnP evolves.

cri-s, 2017-08-25:
Follow the example of OpenPnp and add the modified  driver/feeder

cri-s, 2017-08-25:
For jog inhibit, one possibility could be set machine Speed to 0/Infinity (not nan please) on init and home script (event) and during pin needle is down.
Led/buttons need to be attached to the second driver.

johngrabner, 2017-08-25:
cri-s: re: Follow the example of OpenPnp and add the modified driver/feeder

I have.  My version is a proper super set of the existing ReferenceDragFeeder.  In other words, if no pin down actuator/sensor is specified and no peel actuator / stepper is specified, it does everything the same as the existing ReferenceDragFeeder.

It passes all my testing. Just need to add documentation and extend automated testing to cover this functionality.

cri-s: re Jog inhibit:

It would be nice to have a solution that would not require people wanting to adopt this functionality to write code. Also, I assume everyone would want this to be non-breaking for existing users. In other words, no change required for those users with auto or no homing function.

I was thinking something like a pin actuator in "Machine Setup" with config flag "restrict head movement to safe regions defined in drag feeder".   What do you think?

cri-s, 2017-08-25:
If no homing event exist, request one or write PR.
I think something like this:
config.getDriver().setMaxFeedRate(Double.NEGATIVE_INFINITY); // Gcode driver
config.getMachine().setProperty("origFeedRate",(Double)1234);
and Homing script:
config.getDriver().setMaxFeedRate((Double)config.getMachine().getProperty("origFeedRate"));
// Gcode driver

INF gives automatically a error on Gcode line either depending on
setup ignoring the command or raising a error.

You could setup a event driver.noJog and driver.Jog as example using
above lines inside the scripts and calling from home and init events
that scripts/events too.

It depends how the Pin is implemented, as example on TVM801 machines
the drag pin is
implemented differently and require working move command.

Further init script should remove the actuator from actuator panels.

Thinking more about this i think it is a big mess.
As you have used smoothie and not arduino for the second board, do it in HW.
If you had using arduino i it is possible to implement it inside arduino code,

The tvm802 have a piezo.
Build a ne555 delay-on circuit for the piezo powered by solenoid line (cables).

If solenoid is on more then 1 sec (as example) sound the buzzer.
I think it is more safe and less trouble as the software workaround.
The same is true with home/enable.
If you use M17/M18 as example for enable/disable motors inside connect
and home sequence, you don't need the modification inside openpnp
software.
Insert a delay G4 after having enabled the motors during homing.

themindfactory, 2017-11-20:
So I have a TVM802A, can I be of any help?

johngrabner, 2017-11-20:
I converted the TVM802B to use Snoothie controllers.

Started with OpenPnP, but decided to code everything from scratch in nodejs javascript plus opencv libraries.

Now it does everything I want, including vision on drag feeder, pause on first component of a design to make sure
cad file is correct, to built in tests.

Currently using for production, so kind of slowed way down at the stage of a working command line interface plus vision windows.

Still need to bolt on a server for web based  control.

Will keep you posted once server is on, then will make public

themindfactory, 2017-11-20:
Nice, its too much work for me to do.... Do you know who else is working on TVM802A/B drivers for Open PNP? Awhile back I had thought there must be others that would like to contribute financially to whom ever takes it on, I never expect anything for free.

cri-s, 2017-11-20:
There existed a driver using the code that you could found on github, and
some members have using that, before replacing the machine.
I`m not aware of anyone using that currently.

themindfactory, 2017-11-20:
I have never seen any driver for it anywhere, I google and find zero..... what am I missing ? :-)

cri-s, 2017-11-20:
probalby, all it`s based on
https://github.com/TVM802/TMV802-Protocol
I have seen 3 different drivers, node.js based, c based , both used
linuxcnc driver on openpnp or direct java implementation.
Sorry, because of hd crash i have lost a copy and source of the last
two mentioned drivers.
Maybe the eevblog forum or some user of it is more up to date as it
started and ended
on that forum, with detailed description of protocol and source code
shared over bitbucket account (with invitation).
Cri.S