openpnp/openpnp #631: CvPipeline must be released after use

vonnieda, 2017-08-29:
There are several places in OpenPnP where CvPipeline is used without being released. This has the potential for a memory leak. To find places where this is happening, just search for references to CvPipeline.process() and see if it is followed by a release(). 

To correct, easiest to just wrap the whole block in a try/finally.

vonnieda, 2017-09-02:
Fixed for ReferenceStripFeeder and ReferenceStripFeederConfigurationWizard. There are still more instances, but they are a bit harder to fix due to the large functions they are used in. Will be best to do a little refactoring first, especially in bottom vision and fiducial finder.

cri-s, 2017-09-04:
Bottom vision and fiducial finder already releases the images.

KarlZeilhofer, 2017-10-17:
this topic seems to be related to this discussion here:
https://groups.google.com/forum/#!topic/openpnp/Z3AGhFKEQAw

@cri-s I couldn't find any .release in the file ReferenceBottomVision.java  
Where do you think it is already done?

cri-s, 2017-10-17:
The pipeline release the images before pipeline run and before loading
a new pipeline.
BottomVision reuses the pipeline(s).

cri-s, 2017-10-17:
I have a single pipeline for each camera and process is syncronized .
As i have understand but i'm totally unsure,  if there are 150
components with 30 different parts, at the end there are 150 pipelines
with images that are never released.
To be honest, i don't know the part lifecycle and pipeline and i
simply have a string attribute on parts that specify the pipeline
indipendently of fiducial pipeline or bottom vision pipeline
or top vision pipeline and having one directory inside configuration
directory where pipelines
are stored and shared between parts.

2017-10-17 11:03 GMT+02:00, Cri S <phone.cri@gmail.com>:

KarlZeilhofer, 2017-10-25:
@cri-s very interesting, your approach. Sounds similar to the implementation of referring pipelines in our HeapFeeder. How did you implement it? I think this would be interesting for all users. 

@vonnieda, could you please clarify, what releasing means exactly?

vonnieda, 2017-10-25:
@KarlZeilhofer CvPipeline has a release method which calls release on any saved Mats. This frees up the memory used by the images that are stored as each stage processes. In general, it's not important to be called, because calling process() releases the previous data, but as @cri-s pointed out, since BottomVision uses a pipeline per part, it's possible this could use up a lot of memory during a job run. We need to change it to release after bottom vision is done. This is a pretty easy change, so I will probably do it right now :)

vonnieda, 2017-10-25:
This is now fixed for all existing cases. See https://github.com/openpnp/openpnp/commit/16ab399fccce65c0d4fbdb9c128417b542911850.

In addition, CvPipeline now implements AutoClosable, so it can be used in try with resources blocks such as:
```
try (CvPipeline pipeline = getPipeline()) {
    pipeline.process(); 
}
```

When used like this, the pipeline will be automatically released after the try block.

closed 2017-10-25 by vonnieda