openpnp/openpnp #646: Camera Pipeline Needs To Drop Frames

vonnieda, 2017-09-13:
Recent changes to the ReferenceStripFeeder have made it clear that the synchronous nature of the current camera pipeline is causing problems. The current pipeline look something like:

Capture -> Transform -> Broadcast -> Repeat...

The main issue is that the Broadcast stage broadcasts the image to each listener synchronously and this blocks the camera from capturing further images. This can cause camera frames to be buffered at the hardware and OS level, and can cause vision operations to fail by receiving old images.

In addition, depending on what is happening at the listener stage, it's possible that the camera pipeline gets backed up continuously causing explosive memory and CPU utilization. In this case, the system will never catch up and all images will be "old" with respect to when the camera saw them.

With this in mind, it's clear we need to the ability to drop out of date frames, and to always be able to capture a "new" frame. The former is important for live streaming and the latter is important for vision operations that call Camera.capture(). The goal should be that whatever image is being displayed or processed is the newest one that has come from the camera, and not necessarily the next one in order of capture. 

References:
* https://groups.google.com/d/msgid/openpnp/MEXPR01MB216876E52C348F055ACC26CC9E6B0%40MEXPR01MB2168.ausprd01.prod.outlook.com?utm_medium=email&utm_source=footer
* https://groups.google.com/d/msgid/openpnp/12fe8a48-f425-4261-ac8f-d45589967f5c%40googlegroups.com?utm_medium=email&utm_source=footer

vonnieda, 2017-09-15:
Spent a little time on this tonight. Found the following:
* The two main cameras, OpenCvCamera and Webcams cannot cause frames to be buffered as far as I can tell. Both read a frame from the camera when a new frame is ready for broadcast or being captured and even when forcing listeners to take seconds per frame, the next frame always contains the latest data. Tested on Mac by putting a 1000ms sleep in the listener and then moving an object under the camera to see if the frame ends up more then 1000ms behind. It does not.

    This probably needs to be tested on Windows too, since the underlying camera capture subsystem may handle buffering of images differently.

* ReferenceStripFeederWizard's default pipeline is quite fast excepting two things: 
    1. settleBeforeCapture on the camera limits the frame rate during auto setup. This makes the feedback seem very slow and for auto setup it is not needed since it's live feedback rather than being captured for vision. Unfortunately there isn't a good way to disable this without either modifying the pipeline dynamically or using a second pipeline for auto setup. Do you have any thoughts @richard-sim?
    2. The canny stage adds about 40ms but as far as I can tell is not needed, since the hough circles call includes it. @richard-sim thoughts?

So, on Mac at least, there isn't really a problem per se. Slow image filters do slow the frame rate, but no out of date images are ever delivered.

cri-s, 2017-09-15:
Just comment on settleBeforeCapture, during autosetup wizard, you could set settleTime to 0 and restore
settletime just after autosetup wizard remove the gui hook on camera display.

trcwm, 2017-10-12:
@vonnieda The OpenPnP-capture lib will automatically drop frames, i.e. it always returns the most recent frame. You probably already know this. :)

vonnieda, 2017-10-12:
Yea, this ticket was filed on bad info. We can close it.

closed 2017-10-12 by vonnieda