openpnp/openpnp #648: Use Fiducial Data For Board Scaling When Possible

vonnieda, 2017-09-14:
Currently OpenPnP uses the fiducial data to calculate a board corner offset and rotation. When a board is not perfectly scaled from the manufacturer this can result in offset errors where the work is further from that corner.

By using three fiducials, we should be able to calculate the scale of the board and apply that when placing components. This will solve issues where boards are not manufactured perfectly, and will also help alleviate some of the error caused by an imperfect machine.

I have not fully mapped out how this would all work, but some thoughts are:
* Move the fiducial vision locations into BoardLocation, similar to how Placed is stored. These will be volatile and not saved in the XML. As a side effect, this will also solve the problem of updating board locations after every fiducial run. 
* When three or more fiducials are available, determine the three that give the greatest X and Y offset to better calculate the scale.
* Update Utils2D.calculateBoardPlacementLocation to use the fiducial data on BoardLocation when available, and gracefully fall back to using two without scaling, or none with just BoardLocation's location when the data is not available

#149 is related to this in that it will make sense to calculate offsets from center rather than the corner when we have that data available.

cri-s, 2017-09-15:
Is there someOne that have immediate need for this feature?
Prerequirement is good camera calibration.
Further it works only for single PCB, not for panelfiducials.

vonnieda, 2017-09-16:
Good guidelines for fiducials: http://www.accutroninc.com/pdf/download/fiducial.pdf

vonnieda, 2017-09-17:
Note to self: Be sure that changing the location of the board invalidates the fiducial locations.

vonnieda, 2017-09-17:
Video showing work so far https://www.youtube.com/watch?v=Plz1_8DFCnI

vonnieda, 2017-09-18:
Proof of concept in https://github.com/openpnp/openpnp/commit/bc4f0e25a0b4d14ea444be93de126b253a8a0116. This works well, but the code itself is a hack. 

Some notes on the PoC:
* Finds three fiducials and uses their coordinates to derive the affine transform.
* Affine transform is stored on the BoardLocation and is used in lieu of the normal offset/rotation calculation.
* Can probably be generalized to handle panel and local fiducials. 
* Given a good quality PCB with low scale/shear error, can be used to calibrate or check the machine's calibration. Scale != 1.0 means steps/mm error, shear != 0 means non-squareness error.

And some thoughts on how to implement this for release:
* I think there's value in storing the fiducial locations on the board location for later use.
* Important to make sure the fiducial locations and any derived data is cleared if the board location changes.
* calculateBoardPlacementLocation could use the three nearest fid locations to derive the transform on the fly, make local fids "just work". 
* Three fids must be non-colinear, meaning that at least two of the three must not share an X or Y coordinate.
* To find the furthest three fids we can use convex hull: https://stackoverflow.com/questions/2276488/efficient-algorithm-for-finding-spheres-farthest-apart-in-large-collection
* I envision just checking all fids (global, local, panel) at start so we have the data during job run.

cri-s, 2017-09-18:
Maybe it could help to have a global Fiducial in order to compensate for step/mm change for
feeder pickup and for uplooking camera.

KarlZeilhofer, 2017-09-18:
I think this feature would be a great improvement. Recently I manufactured a PCB with 180x100mm, where I found out, that I have to readjust the steps/mm. Before I hoped, this scaling is already implemented - but it wasn't. Glad to see here some progress!

KarlZeilhofer, 2017-10-14:
just tried it out - works like a charm!

KarlZeilhofer, 2017-10-18:
2 Things: 
1. Should this patch also work for panelization?
2. Could it be, that the corrected coordinate system  is also applied to the bottom vision? This shouldn't be the case, but I had the impression, it does that.

vonnieda, 2017-10-18:
@KarlZeilhofer:

1. No, currently does not work for panelization. It will before it is released.
2. Yes, it's possible if you are using pre-rotate it will mess things up. I'll have to consider that when I get back to this code.

Phara0h, 2018-07-20:
Has this PR died? This is a sweet feature that I want very much.

vonnieda, 2018-07-20:
@Phara0h There was never a PR for it. I wrote a little proof of concept linked above, but someone (or me) needs to take the idea and turn it into an actual feature. My code was just a quick hack to see if it worked, and it did, but it was a hack that would have broken most installations.

KarlZeilhofer, 2018-07-21:
We also would really love this feature! Currently we do the transformation in the placement file with python. 
Further more we have some "nonlinearities" in our mechanics, which would need another transformation. Perhaps one day we will implement it with an automatic selfcalibration.

vonnieda, 2019-02-16:
Paste of IRC conversation on a method that might be better than the one I'm about to commit:

[13:46:42]  <+Nutter>	@vonnieda : re picking best fids - probably the most correct way would be by computing the eigenvectors/eigenvalues through PCA (principle component analysis), as that'll give you an orthogonal vectors for the best-fit oriented bounding rect - i.e. the two vectors that will give you fids with the best results
[13:47:03]  <+Nutter>	it's common enough that there must be a good java lib out there for it
[13:49:55]  <+Nutter>	https://stackoverflow.com/questions/10604507/pca-implementation-in-java - personally I like the simplicity/clarity of #3, Apache Commons Math
[13:56:55]  <+Nutter>	once you have the eigenvectors, it'd be a matter of projecting the fids onto the EV's and picking the 3 that are closest to the corners
[14:16:50]  <@LA3PNA_>	Nutter, wouldn't that require a lot of more computational power?
[14:17:12]  <@LA3PNA_>	I like the idea of using eigenvectors tho
[14:17:30]  <+Nutter>	nothing compared to running the rest of OpenPnP like the vision 😉
[14:18:07]  <+Nutter>	it's really not that heavy an algorithm, and is the mathematically correct way
[14:21:54]  <+Nutter>	for a ballpark, I'd expect it to run <1ms for 100 points, probably far less
[14:22:44]  <+Nutter>	PCA is used heavily in image compression, which has millions of points for any decent sized image
[14:24:12]  <@LA3PNA_>	I am familiar with the math, but not its applications as such
[15:20:38]  <@vonnieda>	Nutter: I don't even know what most of those words mean, so I am going to bookmark that and consider it for another day 😊
[15:21:17]  <@vonnieda>	My naive algorithm seems to be doing a decent job, so I'm going to get it committed and then maybe you could help me understand how to apply what you've mentioned here.

vonnieda, 2019-02-18:
Fixed in https://github.com/openpnp/openpnp/commit/eb2ae16d13be741e2457d8fcb2c710b8138a305f

closed 2019-02-18 by vonnieda

aneox, 2019-02-18:
@vonnieda Hi! So now this feature is enabled by default? No checkbox needed? Good job! Great feature! Thank you!

vonnieda, 2019-02-18:
Correct, no checkbox needed :)