openpnp/openpnp #649: Camera resolution if applying transformation

mgrl, 2017-09-20:
# Problem Report
## Describe the Issue
It seems if rotating the camera there is some math off. Rotating the camera view by 180 degrees, the resolution of the camera is different from the resolution if not rotating. see attached files.

## Steps to Reproduce
1. Start OpenPnP
2. Go to camera properties
3. set rotation to 180°
4. see the information in camera view and compare to the information given if rotation =0

## Expected Result
resolution should not change. 
[image]

## Actual Result
It seems there are black borders added to the images delivered from the camera.

[image]

snapshot:
[image]

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Added further logs which show the camera opening procedure.
[image]
Two things I would like to point out:
1: When opening a camera there are two logs to that. has this to do with some buffered cameras internally used or is there a duplicate of camera stream?
2: the resolution set and reported is wrong in the first time, the camera is opened, the second time its correct.

maybe thats all not a problem, but wanted to let you know...

cri-s, 2017-09-21:
Rotation is for correcting alignments errors, example 3.456 degree. 
For 180 degree you should use both flip options.
640x480 resolution is displayed when the camera don't give out invalid image.
That happens when camera starts, the first 10-15 images are wrong or most of it are wrong and when getting that wrong image camera dimension change in your case. That is normal.

mgrl, 2017-09-22:
OK, just flipping works for me.

commit eb2ae16d (2019-02-18): Compltes the affine transform fiducial system.
Fixes #791
Fixes #649
commit 72d7a9fe (2019-02-18): Merge branch 'feature/fiducial-transforms' into develop

* feature/fiducial-transforms:
  Compltes the affine transform fiducial system. Fixes #791 Fixes #649
  This file was previously changed in this branch to support a new option, but I'm not going to expose that option, so reverting this.
  Remove test code no longer needed.
  Playing around with affine transforms for handling scale and shear in boards using 3 fiducials. https://github.com/openpnp/openpnp/issues/648
closed 2017-09-22 by mgrl