openpnp/openpnp #650: Problem Loading Configuration. Reason was NULL

Norxcode, 2017-09-22:
If you delete all settings in the Machine Settings, Heads, ReferenceHead you receive a error message starting OpenPNP

Configuration Load Error
There was a problem loading the configuration. The reason was: NULL
Please check your configuration files and and try again.

## Steps to Reproduce
1. Start OpenPNP
2. Delete Camera and Nozzle 
3. Close OpenPNP
4. Start OpenPNP

vonnieda, 2017-09-22:
Verified. Here's the exception:

```
2017-09-22 17:16:42 SystemLogger ERROR: java.lang.NullPointerException
2017-09-22 17:16:42 SystemLogger ERROR: 	at org.openpnp.gui.MachineControlsPanel$6.configurationComplete(MachineControlsPanel.java:405)
2017-09-22 17:16:42 SystemLogger ERROR: 	at org.openpnp.model.Configuration.load(Configuration.java:285)
2017-09-22 17:16:42 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame.<init>(MainFrame.java:538)
2017-09-22 17:16:42 SystemLogger ERROR: 	at org.openpnp.Main$1.run(Main.java:150)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(EventQueue.java:97)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:709)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:703)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2017-09-22 17:16:42 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2017-09-22 17:16:42 MessageBoxes DEBUG: Configuration Load Error: There was a problem loading the configuration. The reason was:<br/><br/>null<br/><br/>Please check your configuration files and try again. They are located at: /Users/jason/.openpnp<br/><br/>If you would like to start with a fresh configuration, just delete the entire directory at the location above.<br/><br/>OpenPnP will now exit.
```

The deal is basically that OpenPnP expects to have at least one tool available (nozzle, camera, actuator or paste dispenser) to show in the UI and when there are none it blows up.

I don't think it's worth a proper fix for this, since that would include rewriting huge swaths of the UI, but we can certainly improve the error message so the user knows what went wrong.

cri-s, 2017-09-24:
I think it could be reasonable to not allows deleting nozzles and heads if there exists only 1 element,
nozzle or head depending  inside the gui "action" code that deletes the nozzle.

vonnieda, 2017-09-24:
Agreed, that would do the trick

commit 1fb4e112 (2019-02-16): Fix #650 by refusing to delete last nozzle.
commit 822352ff (2019-02-18): Merge branch 'develop' into feature/fiducial-transforms

* develop: (280 commits)
  Fix #650 by refusing to delete last nozzle.
  Lint
  Fixes #676 by making long lists of actuators wrap.
  Added new variables to GcodeDriver to allow for direction signalling.
  Added XY soft limits.
  * Park XY now always parks the head at the same location, regardless of what tool is selected in   the jog control dropdown. This ensures that if you choose a park location with one tool selected   and then attempt to park with a different tool selected you don't crash the head. This is   primarily a safety impro [... cut]
closed 2019-02-16 by vonnieda