openpnp/openpnp #652: Full path of board filename in job

ticso24, 2017-09-24:
# Feature Request
## Describe the Feature
I like to have a way to view the full pathname of a boardfile within the loaded job.

## How Will It Look?
Always showing the full pathname in the job list is likely a bad idea, but showing it as tooltip on the board filename would be nice.

## Why Do We Need It?
1. What problem does it solve?
Sometimes filenames are not all that informative without a path and so far the only way to lookup the complete path is by looking into the job.xml.
2. Is it useful for everyone, or does it only solve a problem for your machine?
It may be specific to the way I work with board files, but will likely useful for a few others as well.

vonnieda, 2017-09-25:
Good suggestion. Should be easy to implement.

vonnieda, 2017-09-28:
Fixed in #653. Thank you @pfried for this nice fix!

closed 2017-09-28 by vonnieda

vanthome, 2017-12-24:
@ticso24 I saw your videos on youtube. Can you share what hardware you used? Is it a DIY build? I'm also interested in building one.

commit 8cc8ccb4 (2017-09-28): Fixes tooltip incorrect for tables after sorting.
Fixes #652
commit f2b76d45 (2017-09-28): Merge branch 'pfried-jobfile_tooltip' into develop

* pfried-jobfile_tooltip:
  Fixes tooltip incorrect for tables after sorting. Fixes #652
  Adding a tooltip displaying the filename of a board within the JobPanel when hovering the first column. Solves #652
commit 26a8f4db (2017-10-01): Merge branch 'develop' into feature/openpnp-capture

* develop:
  Fixes tooltip incorrect for tables after sorting. Fixes #652
  Adding a tooltip displaying the filename of a board within the JobPanel when hovering the first column. Solves #652