openpnp/openpnp #655: Jog with both cameras (up and down) displayed

Norxcode, 2017-10-04:
# Problem Report
## Describe the Issue
If I have two Cameras displayed on the Console (up and down). and by error make a Move with the mouse in the bottom Window. Head moves to endstop.  
Same happens if you display bottom Camera only. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Jog with the Mouse in the Bottom Camera view. 
2. Head moves to endstop or thru it outside working area. 

## Expected Result
Move head as you jog it with the Down Camera, or an message not allowing jogging as Head might crash into something.

vonnieda, 2017-10-04:
Jogging in the bottom camera moves to the position of the camera. This is
expected behavior. For instance, if you click in the middle of the bottom
camera the nozzle should move to be visible at the center of the camera. If
it doesn’t, please check your cameras’s location and units per pixel
settings.

Norxcode, 2017-10-05:
Thank you for your response.

I must say that I am a bit embarrassed now. It was me changing camera to OpePnPCaptureCamera from WebCamera. To many Changes at the same time. Upgrading to new version, changing camera, trying to get it up and running on same HUB as other camera.

All working good.

closed 2017-10-05 by Norxcode