openpnp/openpnp #658: homing-fiducial-location used from sub-driver

KarlZeilhofer, 2017-10-06:
# Problem Report
## Describe the Issue
Until yesterday we used only one main G-Code driver. Then we installed a pressure sensor, using an Arduino Nano via USB-TTY, using a subdriver to read out its values. 
We have been using a non-zero value for the fiducial home. 
```<homing-fiducial-location units="Millimeters" x="92.34" y="0.23" z="0.0" rotation="0.0"/>```

After setting up the new sub-driver, this position got overridden by the sub-driver and it was by default set to zero, which lead to a limit-switch "crash". 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Use only one main G-Code driver
2. set in the machine.xml a position with a positive x-value, e.g. 100mm
3. test fiducial homing is working
4. restart OpenPNP (optional ???)
5. setup a sub-driver
6. restart OpenPNP
7. Do the fiducial homing

## Expected Result
The fiducial home position should be used from the main G-Code driver

## Actual Result
The fiducial home position of the sub-driver is used

## Workaround
Set the desired fiducial home position also in the sub-driver.  
We are using the develop version from yesterday, commit 77303227e2e80fceb898b1e54d2f25d5dae3718d

vonnieda, 2017-10-06:
Good catch. This really should have never been in the driver in the first place, it should be on the head.

vonnieda, 2017-10-27:
This is more complicated than it seems. If we change it so that only the main driver runs fiducial homing we may leave out axes that are defined on the sub-driver. I think to really solve this we need to add a new option on the driver to enable or disable visual homing for that driver.

vonnieda, 2017-10-27:
Added an option.

closed 2017-10-27 by vonnieda

mgrl, 2017-10-27:
Ahhhh, was wondering, why the homing was performed twice on my machine. I have two drivers (main+sub). Now with the new checkbox having homing only checked on the main driver, the visual homing is performed only once.
Just wanted leave a note if anybody in the future wonders why homing happens twice... That's not needed any more :)

KarlZeilhofer, 2017-10-27:
Ahhh, mine did it 3 times. I thought, this must be like that, to ensure consistency and increase acurracy. Great when this process now is faster :)

commit 1b767e65 (2017-10-27): Added a checkbox to enable visual homing on GcodeDriver. Fixes #658.