openpnp/openpnp #659: Camera Event not in sync with taking a picture from camera?

mgrl, 2017-10-08:
I would like to use the camera events to turn on the lighting of the camera currently used. The scripts are as following and working correct in principle, that is I can see the lights turning on/off as intended.
BEFORE:
 var UplookingCamLighting = machine.getActuatorByName("UplookingCamLighting");
 var DownlookingCamLighting = machine.getActuatorByName("DownlookingCamLighting");

if (camera.looking == Packages.org.openpnp.spi.Camera.Looking.Up) {
	if (DownlookingCamLighting) {
    	    DownlookingCamLighting.actuate(false);
	}
	if (UplookingCamLighting) {
	    UplookingCamLighting.actuate(true);
	}

} else if (camera.looking == Packages.org.openpnp.spi.Camera.Looking.Down) {
	if (UplookingCamLighting) {
	    UplookingCamLighting.actuate(false);
	}
	if (DownlookingCamLighting) {
	    DownlookingCamLighting.actuate(true);
	}
}

AFTER:
 var UplookingCamLighting = machine.getActuatorByName("UplookingCamLighting");
 var DownlookingCamLighting = machine.getActuatorByName("DownlookingCamLighting");
 if (UplookingCamLighting) {
	UplookingCamLighting.actuate(false);
 }
 if (DownlookingCamLighting) {
	DownlookingCamLighting.actuate(false); 
 }

Unfortunately it seems, the light is not on, when the camera actually captures a picture, so the images are just black. 

## Expected Result
Light/Cam should be in sync to have the field of view lighted.

## Actual Result
black/dark image from camera.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
ELP-camera, latest OpenPnP version. Might it be issued by the new camera library which I use already?
Exposure of camera is set to manual and works correct if having illumination turned on all the time.
Would it be possible to turn on the lights a little earlier and turn them off a little later? could not find information regarding to this in the mailinglist.

cri-s, 2017-10-08:
It works, but only if you are using manual exposure and not automatic exposure with manual control.
Camera have both possibility, i don't know.
If you really have manual exposure, you need to configure the camera delay as
Thread.sleep(150)  on begin script and set the configured value to 0. Here i have supposed 150, you
need to use correct values.
Vision.PartAlignment.Before/After is the other script that get activated some time before capturing with
camera and the other people use to switch lights. Maybe it's explained in some Videos.

mgrl, 2017-10-09:
So would it be worth to include a SettleBeforeCapture-Event? See #660. That works well for me...

mgrl, 2017-10-11:
implemented the new event, closed

closed 2017-10-11 by mgrl