openpnp/openpnp #663: Placed Flag isn't set any more

KarlZeilhofer, 2017-10-16:
# Problem Report
## Describe the Issue
The placed-flag was working properly, until a special point in time. But now it doesn't any more.
Actual behaveour: the left column (References) gets colored from yellow over blue to green - this is as expected.
But the checkboxes all stay unchecked when processing the job. 
When I do check them manually, they get saved and restored as expected,
when closing and reloading the job. Also the job processor skips that parts which were loaded with a checked placed-flag - again as expected.

What I am aware of what is different to before:
* I've merged Jason's patch regarding the affine transform for the board area
* added fiducials to the board (which I added manually to the board's parts list)
* marked several parts of the board to be ignored

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Load a job
3. Start the Job

## Expected Result
The checkboxes should be checked one after the other (according to the green state of the referenc-column)

## Actual Result
the checkboxes don't get checked

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
If it helps, I could provide some additional files - please tell me.

mgrl, 2017-10-21:
Have that issue, too. Was working first, but in the second run of the same job the placed flag was not set any more. Will try to find steps to reproduce.

vonnieda, 2017-10-25:
I've seen this happen sometimes and not happen other times. I'm not sure what's causing it to be intermittent.

KarlZeilhofer, 2017-10-25:
Any progress here?
Could it be that it is something that is written into the board or job file?

vonnieda, 2017-10-25:
I am not going to have time to look at this for quite a while. Perhaps @sp-apertus can take a look. They originally implemented the feature in https://github.com/openpnp/openpnp/pull/626.

sp-apertus, 2017-10-26:
I can take a look but so far I have not managed to reproduce this bug. Can you confirm it also happens with the current latest origin/develop build?

vonnieda, 2017-10-27:
Okay, I took a quick look at this tonight and it appears to be operating as expected. I did notice something that may be the cause of the problem, though. The placed flag is set per board, not for all boards. So, if you have a board selected and the placed flag is set for another board, you won't see that until you select the other board. In any case, by the end of the job, the placed flag should be set for all enabled boards.

@KarlZeilhofer Can you confirm this is still happening and that it's not as I described above? A video might go a long way to showing the problem. If it's still happening, can you send me your board and job files?

KarlZeilhofer, 2017-10-27:
@vonnieda and @sp-apertus 
I have now a little test-job, where the bug can be reproducable "siwtched on and off" :)
I came to the conclusion, that the auto-check-placed works, when the checkbox "Check Fids?" for the current board is disabled. 
When this checkbox is enabled, the placed checkboxes are not set any more. 

[issue663.zip](https://github.com/openpnp/openpnp/files/1421721/issue663.zip)

In case you want to try the printed test board:
Fid-Laserjet-1mm Fiducial Pipeline:
```
<cv-pipeline>
   <stages>
      <cv-stage class="org.openpnp.vision.pipeline.stages.CreateFootprintTemplateImage" name="template" enabled="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="debug_template" enabled="true" prefix="fidloc_template_" suffix=".png"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="template_gray" enabled="true" conversion="Bgr2Gray"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageCapture" name="image" enabled="true" settle-first="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="image_gray" enabled="true" conversion="Bgr2Gray"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.Threshold" name="1" enabled="true" threshold="150" auto="false" invert="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="0" enabled="true" diameter="100"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="debug_original" enabled="true" prefix="fidloc_original_" suffix=".png"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MatchTemplate" name="match_template" enabled="true" template-stage-name="template_gray" threshold="0.1" corr="0.9" normalize="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRecall" name="recall_image" enabled="true" image-stage-name="image"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawTemplateMatches" name="draw_matches" enabled="true" template-matches-stage-name="match_template">
         <color r="226" g="42" b="42" a="255"/>
      </cv-stage>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertModelToKeyPoints" name="results" enabled="true" model-stage-name="match_template"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="draw_keypoints" enabled="true" key-points-stage-name="results">
         <color r="12" g="2" b="2" a="255"/>
      </cv-stage>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="debug_results" enabled="true" prefix="fidloc_results_" suffix=".png"/>
   </stages>
</cv-pipeline>
```
And here the package:
```
<package version="1.1" id="FID_1mm">
   <footprint units="Millimeters" body-width="0.0" body-height="0.0">
      <pad name="1" x="0.0" y="0.0" width="1.0" height="1.0" rotation="0.0" roundness="100.0"/>
   </footprint>
</package>
```
Video is currently uploading to youtube...

KarlZeilhofer, 2017-10-27:
@vonnieda your explanation about different boards is correct, but it doesn't get checked in any board.

KarlZeilhofer, 2017-10-27:
Here the demo video of the bug:
https://youtu.be/iPP6QczKWAo

vonnieda, 2017-10-27:
Very strange - thanks for the additional information.

vonnieda, 2017-10-27:
Okay, have tested with your data and it's still working fine for me, with fids enabled and disabled. Can you please send me your machine.xml, parts.xml and packages.xml? I want to try to completely replicate your environment.

vonnieda, 2017-10-27:
@KarlZeilhofer Another option - since this is happening on your machine, and you have worked with the code, can you try to fix the issue and submit a PR?

mgrl, 2017-10-27:
Can confirm that the findings from Karl occur in my setup, too. Check fids off: Placed is set. Fid-Check on: No placed flag. I'm on the current development tree.

vonnieda, 2017-10-27:
@mgrl I've tried to reproduce this several times now and have not been able to. Can you see if you can find the cause?

mgrl, 2017-10-27:
what would be best to do? set a breakpoint near the "placed routine" while picking and placing?
think it's way too deep in the code for me.

but just one shoot, while scanning the code: there is a function getFiducialCompensatedBoardLocation(...) called from the doPlace function in the jobprocessor. Within the function the boardLocation is copied into a new object which is returned and the place function uses that as board location from now on. setting the place-flag seems to be the only feature relying on the boardLocation... maybe there are some internals messed up while copying?

vonnieda, 2017-10-27:
@mgrl That gave me the hint I needed to reproduce this. Working on a fix. Thanks!

PR #691 (2018-01-04): Mark placed again
# Description
fixes #663: if having fiducials checked, placed flag is set correctly.
don't know whether it's an naive approach or it breaks something else? but at least it fixes the bug ;)

# Justification
it's annoying to have the placement flags not set properly.

# Instructions for Use
use as expected, no matter if with or without checking.

# Implementation Details
1. How did you test the change? Be descriptive. Untested code will not be accepted.
on my machine in a testrun
2. Did you follow the [coding style](https://github.com/openpnp/openpnp/wiki/Developers-Guide#codin [... cut]
commit 6ca421d0 (2018-01-04): fixes #663 (placed flag is set again even if fiducial checking enabled)
commit 93739111 (2019-05-19): Merge branch 'develop'

* develop: (162 commits)
  force setting step complete = true if ignoring an error and keep FSM in the same state.
  +moved boardLocationFiducialOverrides as local variable into BoardLocation +getLocationFiducialOverrides returns location if no fiducial check was performed or locationFiducialOverrides if present +functions which used boardLocationFiducialOverrides (Utils2D/calculateBoardPlacementLocation -> calculateFiducialCompensatedBoardPlacementLocation and ReferenceBottomVision/calculateBoardPlacementLocation -> calculateFiducialCompensatedBoardPlacementLocation) b [... cut]
closed 2018-01-04 by vonnieda