openpnp/openpnp #664: Pick and Place Tab's "go to feeder" button shows wrong feeder

KarlZeilhofer, 2017-10-16:
# Problem Report
## Describe the Issue
That button should show an enabled feeder, but it shows "any" feeder for the selected part. 

## Steps to Reproduce
1. Start OpenPnP
2. Load a job
3. Setup 2 feeders for the same part, and enable only the second one of them
4. select in the pnp-tab a placement with the part from the feeders of step 3
5. click on the button "edit associated feeder"

## Expected Result
GUI should jump to the enabled feeder instead

## Actual Result
the GUI jumps to the first feeder - which is disabled

KarlZeilhofer, 2017-10-20:
A even better solution than going to an enabled feeder, could be to set the search term, and find therefore all feeders for that part. In that list the first enabled feeder should then be selected.

KarlZeilhofer, 2017-10-27:
My suggestion for the "even better" solution above had a background. 
If in the search field still is some text, the list isn't complete, and there is a big chance, that the feeder, which should be shown to the user, is not visible and therefore not selected. 

Should I file this under a new bug?

vonnieda, 2017-10-27:
No need, I'll reopen this.

vonnieda, 2017-10-27:
Search is now cleared when using this function. I considered setting the search term to the part ID but I think this would cause confusion for users who don't use the search very often.

commit 58566b3b (2017-10-27): Fixes feeder edit button selecting disabled feeders. Now will prefer enabled but fall back to disabled. Fixes #664.
closed 2017-10-27 by vonnieda