openpnp/openpnp #668: Play Sound, when job is interrupted

KarlZeilhofer, 2017-10-22:
# Feature Request
## Describe the Feature
When a job is interrupted and asks for user input, it should play some sound to acoustically notify the operator. 

## How Will It Look?
To keep it simple, i would provide a default sound deployed with OpenPnP. If the user provides an other file in his .openpnp folder, this file would be preferred.

## Why Do We Need It?
1. What problem does it solve?
In our lab we also have 3D printer, which makes similar noise as our LitePlacer. If PnP stops, one cannot immediately nitice that. 
2. Is it useful for everyone, or does it only solve a problem for your machine?
Notifications should be helpful for any user.

vonnieda, 2017-10-22:
We do have the Signaler and SoundSignaler system, so it could be used for
this very easily.

KarlZeilhofer, 2017-10-27:
Oh, I wasn't aware of that. I think it already does what I want. 
Though, it would be nice, if we can place files under .openpnp/sounds
which override the default sounds.

pfried, 2017-12-06:
Good idea, I will setup a pull request where you can override the error.wav and success.wav files

closed 2017-12-21 by vonnieda