openpnp/openpnp #672: Placements Panel's Status Column doesn't Update immediatly

KarlZeilhofer, 2017-10-24:
# Problem Report
## Describe the Issue
If the status changes, for example due to a disabled feeder, the cell should show "missing feeder" in red colour. Unfortunately this happens only after a refresh of the GUI. E.g. by scrolling up and down again. 

## Steps to Reproduce
1. Start OpenPnP
2. Start a Job. 
3. Select the board, to see the placements in the tab below. 
3. Wait for a disabled feeder

## Expected Result
It should show "missing feeder"

## Actual Result
The status column still shows "Ready"

KarlZeilhofer, 2017-10-29:
Same applies for the enabled check box in the feeders tab/list

tyllurius, 2020-02-03:
Same applies for the enabled checkboxes for panelized boards.