openpnp/openpnp #673: fiducial pipeline setup problems (uses old pipeline).

cri-s, 2017-10-24:
# Problem Report
## Describe the Issue
If defining a Fiducial Pipeline, it use the old pipeline seen from the debug images, as new pipeline
used HoughCircles and old Pipeline Templatematch.
Everything seems looking good, but when pressing manual fiducial check, it don't work and the
debug images make it evident that the old pipeline is used. 
After reset-all pipelines it works. As this was a first setup, this could be done, but if already pipelines
was setup, thsi don't worked.

## Expected Result
Currently edited pipeline get used

## Actual Result
Old pipeline was used, probably a restart worked, but this problem have required some time to 
understand the reason.

vonnieda, 2017-10-24:
@cri-s Were you editing the pipeline in Machine Setup -> Vision -> Fiducial Locator or in Parts -> ReferenceFiducialLocator?

vonnieda, 2017-10-25:
Tested this and it appears to be working correctly. I tested like this:

1. Start OpenPnP with empty config, creating default config.
2. Go to Parts -> FIDUCIAL-1X2-FIDUCIAL-1X2 -> ReferenceFiducialLocator.
3. Click Pipeline -> Edit.
4. Change the name of the "results" stage to "results1".
5. Click Job -> Fiducial Check. 
6. Note that the check fails because no result is found.
7. Go back and change the name back to "results".
8. Run Fiducial Check again, and it works.

If you are able to reproduce this again, please tell me how to reproduce it step by step, or include a video capture showing the problem.

cri-s, 2017-10-25:
Tested today on fresh source and got the same result as you. Instead of editing pipeline it was pasted from clipboard as i did the same when i run into this problem.
Unfortunatly i had no time to collect info about this  and i have no access to this machine anymore.

closed 2017-10-25 by cri-s