openpnp/openpnp #674: Command and Response are Non-Causal

KarlZeilhofer, 2017-10-25:
# Problem Report
## Describe the Issue
When sending a command in the GcodeDriver, we wait for a ```MOVE_TO_COMPLETE_REGEX```. 
But the responses, that are evaluated, could come from a time before the actual command was sent. 

## Expected Result
The input queue for the data input stream should be cleared before sending the new command to the serial device to ensure a guaranteed causal behaviour. 

## Actual Result
Exactly the opposite happens:
```
GcodeDriver.java
...
    public List<String> sendCommand(String command, long timeout) throws Exception {
        List<String> responses = new ArrayList<>();

        // Read any responses that might be queued up so that when we wait
        // for a response to a command we actually wait for the one we expect.
        responseQueue.drainTo(responses);
...
```
All responses received before sending the command are "drained" into the ArrayList ```responses```. 

The consequence is, that e.g. a bottom vision takes place, before the part reaches it's intended z-level. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
It is related to the problem mentioned here:  
https://groups.google.com/forum/#!topic/openpnp/7IF0e8nfNdQ

at the google group there is also a short log-file attached:  
https://groups.google.com/group/openpnp/attach/d9f096c43313/2017-10-19%2016:41:39%20Overlapping%20Moveto.txt?part=0.1&authuser=0&view=1

cri-s, 2017-10-25:
You have 3 possibilities.
Modify gcode that it works, Hard.
Use tinygDriver, Easy, you loose homing dot, backlash and squareness correction.
Use script run stage and script to check the coordinates before returning as first stage. Simple fix that works if implemented correctly.

vonnieda, 2017-10-25:
@KarlZeilhofer This code was written when the driver was quite a bit simpler, and was added in an attempt to guarantee that no responses get dropped.

Currently we read responses until we get ok or an error, then we read any additional responses and then return. The problem is that if there are additional associated responses they are likely to be missed during the last read as they may not have been fully delivered by the slow serial port.

I'm sure I added this for a reason, but now I can't think of a specific example of where this might come up. 

I'd support making this change, but first I think we need to think through the common use cases, primarily around Smoothie, where this might break.

KarlZeilhofer, 2017-10-25:
We are now running TinyG with a replyQeueue.clear() bevore calling sendCommand(...) in the moveTo(...).
Until now without any problems, some 100s part placed...

KarlZeilhofer, 2017-10-25:
What about backing up the data in the queue, send the command, parse the responses, and fuse the remaining data with the backup?
Then nothing gets lost and the response is definitely causal to the command.

vonnieda, 2017-10-25:
@KarlZeilhofer That would involve moving the MOVE_TO logic into sendCommand, which I don't want to do. Better, I think, is to solve the root problem which is that ideally we should not be processing responses from command A while processing command B. I need to spend some time mapping out request/response for Smoothie and make sure this won't break anything.

markmaker, 2021-04-25:
This is addressed by using new TinyG firmware and proper M400 usage. 

All about this, including the caveats and risks of upgrading a TinyG firmware in general, here:
https://makr.zone/tinyg-new-g-code-commands-for-openpnp-use/577/

After flashing new firmware [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions) can configure TinyG commands and options accordingly.

closed 2021-04-25 by vonnieda

KarlZeilhofer, 2021-04-25:
I really like bug trackers! Some bugs need some time, but it is very likely, that they get fixed sooner or later! Thanks!