openpnp/openpnp #675: GcodeDriver: Improve Delay when Waiting for MOVE_TO_COMPLETE_REGEX

KarlZeilhofer, 2017-10-26:
# Problem Report
## Describe the Issue
As discussed here  
https://groups.google.com/forum/#!topic/openpnp/9j2GZpZdTDg  
the delay of 250ms in the method ```moveTo()``` is rather high for a lot of short move commands. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Write a simple method, which calls the moveTo() in a loop with a small movement, e.g. 1mm each cycle. 
2. Run that method

## Expected Result
Fast frequent moves due to a immediate return, when the reply from the controller was received. 

## Actual Result
We evaluate the response with a delay of up to 250ms. That is, why that frequent short moves are rather slow.

cri-s, 2017-11-07:
I have used recently tinyg and had always received the tinyg response for move complete regex. Why you run into timeout?

markmaker, 2021-04-25:
This is made obsolete by using new TinyG firmware and proper M400 usage. 

All about this, including the caveats and risks of upgrading a TinyG firmware in general, here:
https://makr.zone/tinyg-new-g-code-commands-for-openpnp-use/577/

After flashing new firmware [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions) can configure TinyG commands and options accordingly.

closed 2021-04-25 by vonnieda