openpnp/openpnp #676: Splitview not editable (any more) by drag and drop

mgrl, 2017-10-31:
# Problem Report
## Describe the Issue
In the past one could move the splitscreen vertical bar by mouse. Noticed, this is not possible any more. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. try to move vertical bar

## Expected Result
Camera view / tabs get scaled according movement

## Actual Result
nothing happens, vertical bar sticks

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

I'm on current version loaded from page. Windows if it matters.

Anybody seeing this issue, too?

vonnieda, 2017-10-31:
@mgrl This is likely due to the large number of actuators you have, and that panel not flowing correctly. Try switching to the Jog tab and see if that helps.

mgrl, 2017-10-31:
@vonnieda No, it's on every tab. Made the screenshot in the acutators tab to see where it may come from.
As workaround I don't use the window in fullscreen. then I make the whole window smaller, which makes the area on the left side smaller, too. afterwards I make the window on the right side bigger again. the jog-tab, etc. remain small then...

vonnieda, 2017-10-31:
Did a quick test and it does seem to be related to the number of actuators, even when the panel flows correctly. Will need to look into this further. Possible that the layout manager for the actuators panel is locked for some reason.

KarlZeilhofer, 2017-11-01:
same here. I've noticed this problem, but I ignored it for now. I still have a goot ratio between left and right pane. I never saw my actuator buttons in 2 or more rows, only a single row since I'm using openpnp. I'm running Linux Mint 18.2.

commit 4af79106 (2019-02-16): Fixes #676 by making long lists of actuators wrap.
commit 822352ff (2019-02-18): Merge branch 'develop' into feature/fiducial-transforms

* develop: (280 commits)
  Fix #650 by refusing to delete last nozzle.
  Lint
  Fixes #676 by making long lists of actuators wrap.
  Added new variables to GcodeDriver to allow for direction signalling.
  Added XY soft limits.
  * Park XY now always parks the head at the same location, regardless of what tool is selected in   the jog control dropdown. This ensures that if you choose a park location with one tool selected   and then attempt to park with a different tool selected you don't crash the head. This is   primarily a safety impro [... cut]
closed 2019-02-16 by vonnieda