openpnp/openpnp #679: AutoFeeder with mechanical tape advancement

netzmark, 2017-11-23:
# Feature Request
## Describe the Feature
There are some older feeders of Yamaha/Philips having the tape advanced by the mechanism lowered down together with the nozzle (within the same move). So that kind of feeder requires an option that allows to activate an actuator after the pick position is achieved by the head, instead of doing this priorly like today (during the ride). 
Actuator must have possibility to be run as many time as much we want to advance the tape. Like one time for 4mm advancement, two times to get 8mm etc.

## How Will It Look?
Two options to choose, depend what you decide is easier to implement: 
- or you can create in the GUI just new type of the feeder, or 
- add some checkbox to the existing ReferenceAutoFeeder realizing function like "run actuator before pick position" / "run actuator at the pick position".

## Why Do We Need It?
1. It will allow to use that type of feeders. Without the option it's impossible at all.
2. Hard to say how popular are these feeders but for sure are much cheaper than newer more advanced and more complicated.

## Provide Examples
I have already added the service of that feeders (thanks for Cri!) to the OpenPnp. It's working and tested in use for everyday. 
Below is the list of changes to the existing ReferenceAutoFeeder.java :

Line 65:     public void feed(Nozzle nozzle) throws Exception {
+		//change: to move the head to the feeder position before actuator running
+              nozzle.moveTo(location.derive(null,null,0.,null));

Line 80:     else {
+            	//change: to click the feeder n-times 
-		actuator.actuate(actuatorValue);
+             actuator.actuate(Double.parseDouble(nozzle.getId()));

br
Marek

netzmark, 2017-11-23:
It's above corrected automaticaly lost part of sense.
line: actuator.actuate(actuatorValue);
should be: //actuator.actuate(actuatorValue);

markmaker, 2021-04-25:
Can be done by #987.

closed 2021-04-25 by vonnieda