openpnp/openpnp #686: Cartesian Product Planner Fails on Large Jobs

vonnieda, 2017-12-09:
The current job planner performs a cartesian product of all the placements, nozzles, and nozzletips to determine the best order to place parts in. For large jobs on machines with more than 1 nozzle this rapidly consumes all CPU and memory and eventually results in a crash. 

More about the specific failure is documented in this message: https://groups.google.com/d/msg/openpnp/4iffqtbPZjk/4cOZanUlAAAJ

In the discussion linked below there are several suggestions on how to resolve this and @cri-s has posted some code. I have not yet evaluated the code or really spent any time trying to figure out a good solution.

My initial thought is that we can probably just do a very naive approach where for each nozzle tip currently loaded we attempt to try to find a part in the remaining placements that we can place without doing a nozzle tip change. If that fails, we can then just move on to the next nozzle tip. Optimizations could include sorting the placements by nozzle tip requirement. The goal will be to find a reasonable placement solution, rather than a perfect one.

From the discussion at: https://groups.google.com/d/msgid/openpnp/fa3a3dc5-ce96-4f43-bcbe-6f806f5b98d6%40googlegroups.com?utm_medium=email&utm_source=footer

netzmark, 2017-12-10:
Making any improvement I think that would be worth to consider to load the calculated plan to some output file and ability to load the plan from that file instead (checkbox on demand) of online automatic calculation. To allow manual management of the pnp order.

vonnieda, 2017-12-10:
@netzmark Since you are here, can you send me your machine.xml? Want to get a better idea of your nozzle and nozzle tip setup. I'm working on this issue today. You can email it to me at jason@vonnieda.org.

netzmark, 2018-01-04:
Jason may I know how it goes, do you expect some success with this soon?
I've been working using CriS planner but after last official updates (mainly missing flags bugfix) it becomes much harder to compile together with new changes.

vonnieda, 2018-01-05:
@netzmark No progress to report. Will report when there is some :) May be worth trying to convince @mgrl to take it on. He's been on a roll recently! :)

netzmark, 2018-01-05:
yes, he's really been, but he seems be overheated after last doings :)

mgrl, 2018-01-12:
hi,
did some research on that and it's indeed a very complicated, extensive problem for such an open and flexible system if you want to consider all aspects (part's height, workload on each nozzle, minimum nozzle tip changes, ...). It's unlikely I can do that in some time.
For documentation I post some paper titles I found and have here, maybe there are some useful ideas in them (sorted by relevance):
- An Efficient Planning Algorithm for Multi-head Surface Mounting Machines using a Genetic Algorithm
- A HIERARCHICAL METHOD TO IMPROVE THE PRODUCTIVITY OF MULTI-HEAD SURFACE MOUNTING MACHINES
- An aggregated optimization model for multi-head SMD placements
- Hierarchical planning for a surface mounting machine placement
- A MULTI-OBJECTIVE OPTIMIZATION APPROACH FOR MULTI-HEAD BEAM-TYPE PLACEMENT MACHINES
- New Models and Heuristics for Component Placement in Printed Circuit Board
Assembly
- A Hierarchical Optimization Met hod in the PCB Assembly for Surface Mounting Machines
- An improved Shuffled Frog-leaping Algorithm to optimize component pick-and-place sequencing optimization problem
- Optimizing the Operation Sequence of aMultihead SurfaceMounting Machine Using a Discrete Particle Swarm Optimization Algorithm

Maybe for the first it would be a good idea to implement one of the planner other people have on their machines here? There could be switched between two planner dynamically in dependence of the job size?

netzmark, 2018-01-12:
Currently I'm using the planner made by Cri which find as really good. The most important adventage over Jason's planner is that it works :-). Looking on my requirements I can't see any disadventages, and it doesn't stop after fiducials reading at all, just run assembling at once without any visible delay - no matter how large/complicated is the job!
I suspect that this planner may work not properly if Nozzles_Auto_changing is disabled. I just didn't tested it this way. But there is few dependencies in the code regarding to the Changer_on so it may be some problem - and then general disadventage as official Openepnp release should be universal for everybody.
This planner has nice adventage that order of placement depend on parts height, the highest parts placed as the last. But to be more usable this threshold parameter should be programmable in UI of Job... so it would be needed to add it. I didn't care to add it because the height fixed in the code is actually enough good for me.
Not sure if it's needed to fight with some dynamical switching. My oppinion is that better to have it manually selected by user.

netzmark, 2018-01-12:
PS. I've never made any comparison of oficial planner with Cri's planner. When the plan is possible to be built by Jason's planner - then maybe this planner is much faster than Cri's, maybe has less nozzles changing etc etc. Hard to say, and probably it just depend on the kind of project.
That's why I think that giving to user an option of manual choose which planner he want's to use is good idea.

cri-s, 2018-01-13:
I have a hierarchy planner with optimisation.
It doesn't interface well with the api.
I still use this api, but there are a lot of known bug/issues because that.
If during assembly including job pause the job table is not changed it
works. Otherwise job stop must be issued and after a certain stages it is
not possible to stop the job anymore, or to say it better stop and restart
with changed job table.
The provided planner had no hierarchy and no local or global optimisation
code.
It have no optimisation for minimizing nozzle changes as the way of
planning do it in addition to nozzle and nozzle tip setup.
It showed that there was no timing problem and as example it is no problem
to implement the algorithm from Jason (as I have understand) using lists
and maps instead of the new stream thing that produce  stalling because of
extreme memory usage as the emitted timing hooks showed real ridiculous low
values on very big jobs.
My understanding of Jason code is this:
If having 8 nozzles, it select the first 8 components from a list sorted by
component height.
Then it assign each component a nozzle.
The nozzle is choosen from the nozzles that can place this part and the
nozzle is selected where it can place the most number of parts.
Probably there is code to check first if nozzletip can handle this
component.

netzmark, 2018-01-13:
About no nozzles changes minimizing, using this your "simplified" planner I didn't noticed some excessive volume of changes when using three nozzles. At least never situation of come back on some nozzle to the tip which was used before the change. Maybe it's accidentaly but rather everything was looking quite optimal.

vonnieda, 2019-02-26:
Probably fixed in https://github.com/openpnp/openpnp/commit/1b2a4577ac07c9f19cc3b4e157b758535cf45b8e but requires some testing. See https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2019-02-26 for info.

closed 2019-05-27 by vonnieda