openpnp/openpnp #687: High CPU and Memory Usage with OpenPnpCaptureCamera

vonnieda, 2017-12-09:
Marek reports very high CPU and memory usage when using the OpenPnpCaptureCamera. Niels investigated the capture library itself and found it to be pretty lean, so likely there is a memory issue in OpenPnP.

A few things to consider:
1. Comparing to the OpenCV driver may be a red herring as the OpenCV driver often arbitrarily limits resolution and FPS. It can be difficult to be sure you are comparing like to like.
2. Need to make sure frames received from the capture library are being freed properly.
3. Read through the posted discussion below and note in particular Marek's post at https://groups.google.com/d/msgid/openpnp/15f96db4-9aee-48f6-a4b7-84b3c3a78926%40googlegroups.com?utm_medium=email&utm_source=footer which contains some useful test information. 

Discussion at: https://groups.google.com/d/msgid/openpnp/8ec181b6-4c5c-4d37-9533-b0445bff248b%40googlegroups.com

trcwm, 2017-12-10:
Note: the DLL mentioned by Marek is qcap.dll, which is part of DirectShow.

netzmark, 2017-12-10:
Since I know that fps value made in settings is not tantamount to the real received settings visible in the "image info" information - I'll make a new comparison with the same real condidions and report here very soon.

cri-s, 2017-12-18:
Using ImageCamera, i was only able to get 15-21 fps when setting fps to 60 .
ImageCamera just returned the image, it don't do nothing else as Imagecamera is the modified version
needed for making tests, if the image is a webcam image, it return that image, otherwise it do the usual
processing extracting only the widht/height from image.
Further i have noticed some memory leaking when setting certains combination of settings.
From this behaviour, i deduce that openpnp cannot more then 10fps on real condition during assembly,
maybe onlye half, because only two . Running assembly lowered the fps 1.5 fps , this without bottom vision enabled. Because of threading, i could run several cameras getting the same fps. 
I don't know if there is some error on displaying the fps or something else.

markmaker, 2018-04-15:
Hi
I just wanted to confirm the high CPU usage issue for the two Full HD ELP cameras I'm using with the 1280x1024 MJPEG setting. Switching to YUV codec (6fps) will resolve the CPU issue but the camera will be extremely sluggish, the scanlines will really be spaced out over 1/6th of a second. Huge rolling shutter effect, very long settling time required.
Here's a screenshot showing where the CPU time is used inside OpenPNP:
[image]
Maybe this helps.

Taking a step back and combining this issue with the [USB bandwidth issue (still requiring you to have two dedicated USB connections to the machine)](https://github.com/openpnp/openpnp/issues/717#issuecomment-381389641), I'd like to ask the question whether it's at all needed to get fulltime "TV coverage" in OpenPNP. It would suffice to take single shots when CV needs them or when the user jogged. So during a job one would see the last relevant picture taken, be it from the down or the up camera, with some time to study it. This would be much more informative than the "rolling shutter show" of the clutter on your machine deck combined with some forlorn blurry view of your ceiling :) Even better there could be a new "virtual" camera showing the last picture taken by _any_ camera, so the view would switch back and forth between relevant views _in full size_. As a side effect it would free up USB bandwidth and lessen CPU requirements. With just still shots required (as a minimum), one could write capture drivers for much simpler (i.e. still) camera setups and connections. There could still be the option to switch capable cameras "live", ideally with some mutual exclusion made sure (switch any other "live" cam off first) to allow for single USB connection + hub setups. 

Just my 2 cents :)

(I'm still struggling much to understand the very cool but highly abstracted model driven code of OpenPNP... hope to get to a point when I could actually offer to help with coding such things).

_Mark

vonnieda, 2019-02-16:
Closing this as fixed by the latest OpenPnpCaptureCamera FPS changes.

closed 2019-02-16 by vonnieda