openpnp/openpnp #690: Update documentation to ReferenceStripFeeder

mgrl, 2017-12-23:
I'd like to update the documentation in order to have some examples to get the feeders angles right when one will setup a job. or maybe make it more general for strip feeder and automatic feeder...

made a simple drawing just for me, but could refine it:
[image]

I found it unintuitive that the 0° direction is opposite of travel direction. But I think the opposite of travel direction is the feed direction then, if one would think of an automatic feeder, which would advance from left to the right... 
is this drawing correct for all circumstances? am I missing something? hopefully it could prevent others from tinkering around with the angles so long as it took me...

mgrl, 2017-12-30:
okay, made two images I would like to see in the wiki for the strip feeder:
first is a drop in replacement to the one currently on the site but with that ugly black background.
[image]
second are some examples, I had luck to setup the feeder with:
[image]

@vonnieda  maybe you would like to take the images over into the wiki? I don't know how to get these there... editing the page itself would be no problem. if changes are needed, let me know

vonnieda, 2018-01-02:
@mgrl These look good to me. Here's how to update the Wiki with images:

1. Upload the images to an Issue comment. You've already done this, but read on. Github lets you drag and drop images into issues, but not into Wikis. When you do this, a unique URL is generated and you can use this URL anywhere to reference the image. You can get the URLs by editing the comment and then copy the URL text. Your two images are:

    `![strip feeder orientation](https://user-images.githubusercontent.com/3868450/34457250-6b69f62a-edab-11e7-8030-0eeed21a9692.png)`

    and

    `![rotation in tape_examples](https://user-images.githubusercontent.com/3868450/34457251-6b83c38e-edab-11e7-8e5a-f6e4212db124.png)`

2. Click Edit on the Wiki page https://github.com/openpnp/openpnp/wiki/ReferenceStripFeeder/_edit and just paste in the above URLs to reference the images. Feel free to replace my black background image with yours above. Yours is much nicer :)

Let me know if you run into any problems making the updates. I could have done it for you, of course, but I figured showing how to do it would be more valuable. 

Please close this issue once you are satisfied with the documentation.

Thanks,
Jason

mgrl, 2018-01-02:
> Github lets you drag and drop images into issues, but not into Wikis. 
Thanks for description, that was the only problem.
Didn't know that having images stored in issues is okay to use them in the wiki then.
Have updated the page.

closed 2018-01-02 by mgrl