openpnp/openpnp #695: Bug in Errorhandling FSM/JobProcessor?

mgrl, 2017-12-28:
# Problem Report
## Describe the Issue
While working on the error handling stuff I stumbled upon an issue I don't know yet where the problem might be. Anyway, two if two error messages follow on each other, the second one is an exception it seems, and there is no ability to handle the error interactively (see below).

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Make sure you have a source of error for sure (eg. raise the vacuum level for a vac sensor that never can be hit or have a fiducial that can't be recognized correctly would be the easiest)
3. Start a job with at least 2 placements to have at least two error messages
4. wait for the first error to occur (in my case vacuum level):
[image]
5. do skip or continue
6. wait for another error to occur. the error now looks like this:
[image]
the machine remains at it's position after clicking OK. the machine can then be paused and continued. afterwards the correct error is presented (like the first image).

## Expected Result
I'd like to see error messages with options to handle them always.

## Actual Result
what I described above. can just click OK.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
please note, that I had these issues already on older development versions, not with the latest changes in it...

hopefully you have an idea what this might be about? maybe with some tips from anybody, I can fix it?

pfried, 2018-02-16:
I did some placing today, I think this is closest to what I am experiencing:

If I do have 2 consecutive mispicks (no part at nozzle) the first can be skipped, the second first shows an error which i can click away (like the vacuum error), then the second skip window comes up.

If I click skip again then the job will continue, but after completing it I am unable to place the missed parts.

So exact same as above but additionally the FSM gets kind of mixed up and if I hit play after the frist run it does not place the missed parts.

If I place a part on the second nozzle while the window is up and hit retry then I can continue without a problem (and place missed parts afterwards). Weird issue, but somehow this is not working correctly.

vonnieda, 2019-05-27:
Fixed in https://github.com/openpnp/openpnp/commit/97e5bfc54827965a24eb710fa0932c3cf46ce29f

closed 2019-05-27 by vonnieda