openpnp/openpnp #701: How can you add Location objects in Python script?

johnthagen, 2018-01-30:
# Problem Report
## Describe the Issue
I am trying to reproduce [Move_Machine.js](https://github.com/openpnp/openpnp/blob/develop/src/main/resources/scripts/Examples/JavaScript/Move_Machine.js) in Python.

The [list of global variables ](https://github.com/openpnp/openpnp/wiki/Scripting#global-variables) doesn't seem to list how one would [Location](https://openpnp.github.io/openpnp/develop/org/openpnp/model/Location.html) into Python.

What I'd ultimately like to do is:

```py
nozzle = machine.defaultHead.defaultNozzle
location = nozzle.location
location = location.add(Location(LengthUnitMillimeters, 10, 0, 0, 0))
```

But I get pop-ups that state that `Location` is not in scope.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Create a Python script in `~/.openpnp/scripts` that tries to import `Location`.
4. Run the script
5. See an error pop up.

## Expected Result
Be able to move the nozzle from a Python script.

## Actual Result
Pop-up errors that `Location` is not in scope.

## (Related feature request)

More Python examples would be great. Reproducing what is in the Javascript examples would be a great start and would probably have avoided this question in the beginning.

vonnieda, 2018-01-31:
See http://www.jython.org/jythonbook/en/1.0/JythonAndJavaIntegration.html for information on Python / Java integration. You should be able to do something like `from org.openpnp.model import Location`. You can see the classes and API at http://openpnp.github.io/openpnp/develop/.

johnthagen, 2018-01-31:
Works, thanks! Would you accept contributions of more Python examples?

vonnieda, 2018-01-31:
Absolutely!

johnthagen, 2018-01-31:
Can be closed by #702 #703 #704 #705 #706 #707

closed 2018-01-31 by vonnieda