openpnp/openpnp #712: SystemLogger NullPointerException

romain145, 2018-03-20:
The log system is not working and shows the following :
`
2018-03-20 21:40:18 SystemLogger ERROR: java.lang.NullPointerException

2018-03-20 21:40:18 SystemLogger ERROR: 	at org.apache.commons.io.FileUtils.copyURLToFile(FileUtils.java:1257)

2018-03-20 21:40:18 SystemLogger ERROR: 	at org.openpnp.Scripting.<init>(Scripting.java:94)

2018-03-20 21:40:18 SystemLogger ERROR: 	at org.openpnp.model.Configuration.load(Configuration.java:288)

2018-03-20 21:40:18 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame.<init>(MainFrame.java:538)

2018-03-20 21:40:18 SystemLogger ERROR: 	at org.openpnp.Main$1.run(Main.java:150)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)

2018-03-20 21:40:18 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(Unknown Source)

2018-03-20 21:40:21 SystemLogger ERROR: log4j:WARN No appenders could be found for logger (com.github.sarxos.webcam.Webcam).

2018-03-20 21:40:21 SystemLogger ERROR: log4j:WARN Please initialize the log4j system properly.

2018-03-20 21:40:21 SystemLogger ERROR: log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
`

The Submit Diagnostics feature returns an error as well:
`
Validation failed (422): Missing required 'user' field.
`

Using openPnP Version 2018-02-21_02-32-09.25a4c60

vonnieda, 2018-03-21:
> org.openpnp.Scripting.(Scripting.java:94)
> org.openpnp.gui.MainFrame.(MainFrame.java:538)

Thanks for the report. I think an example file was not checked in properly.
Will look into that.

The logging error is expected, and does not indicate that logging is not
working in general. The WebcamCamera module uses a different, unsupported,
logging system so it's just that component that is not logging correctly.
This can be safely ignored.

As for the Submit Diagnostics, it looks like something might have changed
in the service those go to. Will check into that as well.

Is this causing the program to not work for some reason?

vonnieda, 2018-03-21:
> Thanks for the FB.
> No it is not causing the program to not work, however it is my first time
> using it, so logging could help ;)
> Can you point me to a known-working revision for the time being please?

The Stable branch from http://openpnp.org/downloads/ should be good.

Just to reiterate though, logging is working fine. It's only failing for
that one module, and that module does not do much logging to begin with.

commit b60d6a00 (2018-04-25): Fix and organize example script names. Fixes #712.
commit ae08caac (2019-05-19): Merge branch 'develop'

* develop:
  Changes the Feeders tab to use the property sheets system, allowing a feeder to have multiple configuration panels. Fixes #737.
  reintroduce multiple uplooking camera support
  ReferencePnpJobProcessor: print offset in debug output
  ReferenceBottomVision: add nicer debug output
  The internal X coordinate set after homing is not compensated for non-squareness after homing to non-zero Y. The non-squared machine X is used directly. This results in incorrect coordinates displayed immediately after homing. When capturing these coordinates and returning to the [... cut]
closed 2018-04-25 by vonnieda