openpnp/openpnp #715: Strip feeder option "Rotation in Tape" is inconsistency with another feeders

aneox, 2018-04-01:
# Problem Report
## Describe the Issue
In strip feeder option "Rotation in Tape" is inconsistency with another feeders. For example in drag or auto feeders we set pick rotation and it works as expected. In strip feeder, if we set 90 degreed Rotation in tape, we expect 90 degree pick rotation, but if strip is rotated 270 on table, vision detect it and pick rotation will be 270+90 = 360 = 0, fully wrong. So we need set Rotation in tape = 0, but in another feeder, with same strip rotation, 90 part rotation works as expected, inconsistency.

I suggest to change concept of "Rotation in tape" for strip feeder, make it just "Pick rotation", as we be used to with drag and auto feeders. Code is easy to fix, we need just made vision correction rotate to nearly 90 or 0 or 180 or 270.. i going to test it and share patch. But i do not know how to continue to support machines, which is already using wrong concept of "Rotation in tape". Or maybe iam wrong?
I think if part on board has rotation 90 and part in strip looks in camera the same rotation, it should have same 90 rotation. It will be easy to understand and setup correct way
## Steps to Reproduce
set strip rotated 270 on table and set Rotation in tape 90

## Expected Result
pick with 90 degree rotation nozzle

## Actual Result
rotation will be 270+90 = 360 = 0, fully wrong

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
I suggest to change concept of "Rotation in tape" for strip feeder, make it just "Pick rotation", as we used to, with drag and auto feeders. 

## How Will It Look?
https://yadi.sk/d/4RpHnBWV3TxYuS
But need to add compatibility for already working machines.

## Why Do We Need It?
It will be easy to understand and setup correct way

mgrl, 2018-05-15:
hi!
From my understanding you should come out like this with the current implementation, don't you?
[image]

At first I found it confusing, too. But how it is currently implemented, it is independent from the feeders orientation on the table. Maybe it's inconsistent because the stripe feeders' rotation could change very likely compared to autofeeder, so it's taken into account only for them, not for the fixed autofeeder?

aneox, 2018-07-02:
i have added my approach in pull request. By default its off, and program will work in old way. To active, you need to set checkbox in machine settings.

vonnieda, 2018-07-07:
I agree - it's inconsistent. I'm not sure, though, if this feeder should be changed, or if all the other ones should be :)

It was done like this because the strip feeder doesn't have a user defined pick location. It just has the locations of the two holes. The pick location is derived from those holes. Since it didn't make sense to have rotation on round holes, I left the Rotation field off and added a Rotation field just for the part.

This is important because many people use Auto Setup to setup the strips, and Auto Setup does not capture a rotation or a pick location - just a hole location.

So, I think the current code is fine and we should not merge the changes in #740, but I am open to discussing this further.

aneox, 2018-07-08:
@vonnieda 
For example, i need to setup diode strip. I do not want spend time on wiki and docs, just move camera to board, look at silk, found anode cathode, look at board placement table part rotation. for example its 90 degrees, so iam going to strip, found that diode in camera looks the same as on board, so i just set same 90 degrees in rotation field. fast and 100% working approach.  Even if diode in strip will be in another rotation, it easy to add or reduce degrees in mind, according to board placement rotation. This approach now works with all feeders, except stripfeeder.

vonnieda, 2018-11-26:
Hi @aneox, I am working on merging #747 and have been looking at this feature. I don't think this feature should be merged. It ties feeder setup to board rotation and board placement rotation which means that your feeder setup has to change for every board that uses the part in a different rotation.

I think this is too specific to your setup and not reflective of many OpenPnP users.

I suggest that you implement this with a simple script, instead. It could work like this:

1. Move camera to **feeder** and rotate the crosshair until pin 1 is up (or down, or left, whatever you prefer). Set this as rotation in tape.
2. Move camera to **placement** and rotate the crosshair until pin 1 is up (or down, or left, whatever you prefer). Run script which subtracts current rotation from rotation in tape.

This should accomplish the same thing, and makes it even fewer steps and no in the head calculations.

What do you think?

closed 2018-11-26 by vonnieda

aneox, 2018-11-26:
> It ties feeder setup to board rotation and board placement rotation which means that your feeder setup has to change for every board that uses the part in a different rotation.

Hi @vonnieda !
Interesting notice, but what about drag feeders and auto feeders in this case? If we rotate board, we should change rotation in these feeders settings also?

vonnieda, 2018-11-26:
@aneox No. The purpose of rotation in every feeder, including the strip feeder, is to tell OpenPnP the orientation of the part within the tape. 

When OpenPnP picks **it first rotates the nozzle** to that rotation, then picks, then rotates back to 0. This ensures that after the part is picked it is always rotated to 0. Then when the part is placed, it is placed at the placement angle + board angle. This keeps the rotation of the feeder and the rotation of the board separate. This also facilitates bottom vision by ensuring the part will always be in the same orientation regardless of feeder.

aneox, 2018-11-26:
@vonnieda but how auto feeder know tape rotation ?))) I thought only strip feeder know tape rotation, because it has holes coordinates and direction of feed.

aneox, 2018-11-27:
@vonnieda Thank you Jason!
In this post https://github.com/openpnp/openpnp/issues/715#issuecomment-403238417
you agreed that autofeeder and stripfeeder rotation works differently and it's inconsistent)

My goal was not to understand how strip feeder pick rotation works now)))
My goal was to make it consistent with other feeders. And easy to understand for new users.

My native language is Russian and maybe i cant fully describe that i mean, sorry.
In my experience with strip feeders i have met troubles, so for me it was easier to make this patch. May be i am wrong and not understand how its works now. Now i have not strip feeders and cant try your approach. I trust you in this decision, so thank you for your patience) Good day!

labamichnetvoll, 2024-04-17:
hi there, 
I've seen this page with good explanations for the strip feeders : [https://github.com/openpnp/openpnp/wiki/ReferenceStripFeeder#rotation-in-tape](url)
Is there something simular for autofeeders? The Wiki page there is almost empty. How does the rotation works there?