openpnp/openpnp #717: Update Camera Wiki

johnthagen, 2018-04-02:
At this link: https://github.com/openpnp/openpnp/wiki/USB-Camera-Troubleshooting-FAQ#why-is-it-a-software-limitation

There are links to two issues which have since been marked closed. Should the wiki be updated to reflect this?

markmaker, 2018-04-15:
Hi
 I just wanted to comment, that the USB limitation still applies to MJPEG cameras. At least with the two ELP Full HD cams I'm using. All the video modes just "fill up" the bandwith. If you choose a lower resolution you atomatically get a higher frame rate and vice versa. There seems no way (yet?) to set the native frame rate, it seems the driver will just throw away surplus frames when you choose a lower frame rate in OpenPNP. BTW: I also suffer from the [high CPU usage issue,](https://github.com/openpnp/openpnp/issues/687) 
_Mark

vonnieda, 2018-04-25:
Updated, thanks!

closed 2018-04-25 by vonnieda

vonnieda, 2018-12-05:
@markmaker Updated Wiki to include stronger language about sticking to the recommendations: https://github.com/openpnp/openpnp/wiki/OpenPnpCaptureCamera#troubleshooting

Do you think this resolves the issue?

markmaker, 2018-12-05:
@vonnieda Certainly an improvement. I'm still on it trying to find specific causes and effects and limits. 

Thanks.