openpnp/openpnp #718: Documentation problem of MinAreaRect for Bottom Vision

kakemoms, 2018-04-10:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
The default pipepline with MinAreaRect does not give consistent part orientation for Bottom Vision. According to [https://github.com/openpnp/openpnp/wiki/Bottom-Vision](url) it should give part orientation that can be used to rotate the part, but as shown is the discussion here: [https://groups.google.com/forum/#!topic/openpnp/VR8UyFN52_c](url) it was not inteded for this.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Select the available feeder.
4. Select pick next part.
5. Move component above bottom vision camera
6. Use part pipeline to identify orientation
7. Rotate bottom camera 90 degrees under camera settings
8. Use part pipeline to identify orientation and get a different result

## Expected Result
Expected the pipeline for bottom vision to give part displacement (x,y) and rotation displacement for use in processing the part and successfully place it on a PCB.

## Actual Result
Part is sometimes rotated 90 degrees from actual PCB pad. This can vary with part size and type.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
See discussion at [https://groups.google.com/forum/#!topic/openpnp/VR8UyFN52_c](url) for images.

The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

cri-s, 2018-04-11:
I have this description of official stages that i have, maybe it helps you for  something. 

Add									ADD TWO IMAGES ( simple Core.add, not really useful )
BlurGaussian						The principal BLUR, kernelsize must be odd , used for noise supression, after manipulate histogram, before doing corner detection, blur + threshold simulate erode 
BlurMedian							Median blur, used sometime as small erode/dilate substitute
BottomVisionCriS					Default bottom vision pipeline as single stage	
ClosestModel						find the rotatedRect from MinAreaRectContours based on prevous stages rotatedrect sizes (templates ???) 
ComposeResult						same as ImageRecall with the possibility of ModelRecall too.
ConvertColor						color conversion codes , hsv full is 255 for H, otherwise it is 180 for H. Most graphic program uses 255 or 360 
ConvertModelToKeyPoints				as name say , keypoint holds position, radius, angle
ConvertModelToPoints				as name say , points only holds position
CreateFootprintTemplateImage		used for fiducial matching , depredicated
CreateModelTemplateImage			Extract  rotatedRect (MinAreaRect or MinAreaRectContours [ first contour ] ) image and replace the current image with the rotatedRect image
DetectCirclesHough					hough circle detection
DetectEdgesCanny					canny  edge detector
DetectEdgesLaplacian				laplacian edge detector
DetectEdgesRobertsCross				roberts cross edge detector
DetectFixedCirclesHough				special stage for stripFeeder wizard or not.
DilateModel							expand or shrink a model by pixels
DrawCircles							draw circles from houghCircles stages
DrawContours						draw contours [blobs]
DrawImageCenter						draws a cross for image center
DrawKeyPoints						draw keypoints
DrawRotatedRects					draw rotatedRects (MinAreaRect...)
DrawTemplateMatches					draw Template match box
FilterContours						filter contours based on pixel area (wrong values for convex contours)
FilterRects							Filter rotatedRects (MinAreaRect...)
FindContours						find Contours, bad implementation
GrabCut								manual grabcut, doubt usefulness
HistogramEqualize					grayscale only, missing color version
ImageCapture						as name says
ImageRead							as name says
ImageRecall							as name says
ImageWrite							as name says
ImageWriteDebug						as name says , only when logging level is set to debug 
MaskCircle							mask the image using circular mask, negative values mena inverse masking
MaskHsv								color filtering
MaskModel							mask out a model or outside the model, if finding the outline of chip, it is capable of masking out the rest of the found results or to blank out not wnated contours.
MaskPolygon							some sort of manual fixed mask of various shapes.
MaskRectangle						mask the image, negative values means inverse masking
MatchPartTemplate					templatematch using rotatedRect (MinAreaRect or MinAreaRectContours [ first contour ] ) testing it against template on 4x 90 degree rotations adjusting rotatedrect angle
MatchTemplate						Templatematch, no good implementation,
MinAreaRect							Group all contours and returns a single MinareaRect . This is the stage required as result for the bottomVision pipeline
MinAreaRectContours					Convert contours to List of MinareaRect, angle is relative to internal values and are not absolute angle values.
Normalize							grayscale = image normalisation, extend image level from 0 to 255 gray levels. On color image, do color segmentation, removing shades and leave only color information.
OrientRotatedRects					tryes to manipolate orientation of detected part (looseFeeder) to have it horizontal or vertical, can be used, bad implementation
ReadModelProperty					read a model from a stage and query model method, not handling null model return, unknow usage
ReadPartTemplateImage	    		implementation of template images, rude, not recommendet to use
Rotate								Rotate a image , extended version accept either degree or stage name, basic version was used for some testing, bad implementation
ScriptRun							run scripting with single argument line.
SetColor							set image to single color, for black maskCircle is usually used, unclear use
SetModel							older version of SetResult, depredicated
SetResult							load/set of image/model , probably used for testing something from java  
SimpleBlobDetector					a lot of setting for simple blob detection, there are some advantage, as automatic grouping, auto thresholding, check opencv simpleblob (2.4) java doc.
Threshold							otsu (auto) and binary or binary_inverse threshold, only 8bit image
ThresholdAdaptive					adaptive threshold
WritePartTemplateImage				conterpart to ReadPartTemplateImage

For the angle of rotatedrect, if you manipulate the angle, expand it to 180 degree from 90 degree, as you
probably have seen, this invalidate the rectangle height/width/location (blue rectangles).
The second issue is that opencv have a different coordinate system as opencv, and as example on 
openpnp the rotation is CCW , on opencv it is CW , the  different coordinate systems implicitly convert
CW to CCW rotation, but if you need using it inside opencv, inside scripts, as example, it is CW.
OpenCv uses Height for longer side of components, OpenPnP uses Withe for the same thing.
If you are interested, this is the best and simplest explaination of the RotedRect angle, height and width.
https://namkeenman.wordpress.com/2015/12/18/open-cv-determine-angle-of-rotatedrect-minarearect/
Mostly RotatedRect  is further processed on other stages, and this is the principal reason why it is not
changed. It was possible to create a own rotatedRect class that replaces rotatedRect and gives out absolute variables, but as feeder base orientation is inconsistent, this is not really possible
without correcting the feeder orientation inconsistency.

cri-s, 2018-04-11:
> What is missing is the step between OpenCV and OpenPNP.
>
> Ok, the namkeenman explains it fairly well. If the rotation is between -90
> to about -45, the component must be rotated in one direction. If it is
> between -45 and 0 it must be rotated in another direction. If one expand
> this from 90 degree to 180, this boundary condition has to be maintained.
> E.g. one has to end up with a rotation which is in the range -45 to 135
> (converted from CW to CCW) to get consistent results. This can (probably) be
> expanded to 360 degrees, e.g to a range of -135 to 225 to give a more
> universal true stage.
>
> What remains is whether this solution would give other problems (to existing
> pipelines).
>
> What do you think?

To expand this from 180 degree to 360 degree, the feeder coordinate
system, and component coordinate system must be deterministic, as
example pin1 is lower left edge
and longer side of component is horizontal side.
This implies that cad data and pcb data need to be different. Most
users want avoid this.
There just want to have different rotation on feeder for IPC-7×51
“Level A” or Level B
component rotation, or orientation system, and don't want to need to
touch the centeroid
data without any abstract (and additional complexity) level.
When doing lean manufacturing, having such a abstraction is a requirement.
Having this abstraction, it is mandatory using external centeroids
editors in order to prepare the pnp data. Annother different Software
to learn for most occasional users.

When doing assembly for others, this additional complexity have a lot
of advantages,
but if doing only inhouse assembly, it is just work that can have some
advantages, but
generally the disadvantage is bigger then the advantages.

That is the big problem when you want have absolute coordinates, as
then you must provide.

Even the coordinate system is different from user to user. One use
-180 to +180, the other
0 to 360, the other -90 to +270 and for the one user, 0 degree is
horizontal, for the other
is vertical.

It is not a problem to introduce a class that hide the internal
rotatedRect and present
other angles and other width and height conventions, the question is
more, if it is wanted
to loose flexibility and time in favor of having such convention, that
allows to share feeders
between jobs with different component orientation system and for this
modify the centeroid data from the layout convention to pnp machine
convention of part rotation ?

 Further, if the returned angle is 42.222 degree, and component should
be mounted on 45 degree, this really helps you in some way. As note,
the mount position is never displayed.
But in reality it is not 42.222 degree but 90 degree off from that.
What is the point to have
such a number and indication that is always relative to how you insert
the component,
and if having the abstraction, beside of sharing the feeder, the gain is little.

kakemoms, 2018-04-11:
Ok maybe I misunderstood the bottom vision document completely, but it states:

"The basic idea for bottom vision is:
Pick a part from a feeder.
Center the nozzle over an Up looking camera and take an image.
Using a CvPipeline, determine the part's offset from center and whether or not it is rotated.
Provide the resulting X, Y and Rotation coordinates to the JobProcessor so that the error can be corrected during placement."

So it says that Rotation is part of the point of bottom vision. Sorry for my incompetence, but if one is not to use MinAreaRect for bottom vision, how is one supposed to image and rotate the part?

cri-s, 2018-04-11:
This is correct, and MinAreaRect is needed, but the "ShowRotation" flag inside
DrawRotatedRect is not for the BottomVision, it have some use inside
loose Part recognition.
Example if you look up leds or diodes or soic from loose feeder that
rotation info is nice to have and it is the pipeline or the
stages/scripts inside the pipeline that detect the orientation of
component. This automatically means that this angle inside that
pipeline is valid.
Instead on bottomVision the angle information is processed outside the
pipeline, only the
pipeline result is used to get the angle/size/location of the part.
The pipeline itself have limited knowledge of the
orientations/offsets. Basically all the calcs are relative to the
placement location or angle 0 (no prerotate) where the placement
inside feeder (and feeder angle) is determinant for the component
rotation orientation.

IIf you want/need sharing feeders between different jobs that have
different part
component orientation rules the pnp machine itself need to have one
single rule for part
orientation. Thus involves usually that the longer (vision recognized)
 side is horizontal
and pin1 is left/botton.  Further it requires that the centeroid
placement files need to be
adapted to the pnp machine orientation. This generally involves
external centeroid tools
that doing this conversion automatically. Every bigger pnp machine (cph > 7000)
does this. The advantage is that feeder can be shared and  that setup
can be controlled
or standarized better resulting in less errors. The disadvantage is
less flexibility and more
time needed to setup things. Sharing feeders gain some time, but if
you make only you'r
standard orientation, there is more work or time needed.
Machines with less then 4500cph  it depends if machine is or can be
part of assembly line,
If not, then mostly components orientation  rules depends on feeder
setup, not on machine
vision rules.

Hope this clearify the concept, otherwise ask again or specific.
Similar  draw center is not the nozzle position on uplooking vision,
it is the point where the nozzle pick up the part on downlooking
vision, and contrary to
"showOrientation" it have some uses on uplooking vision.too.

closed 2018-04-12 by kakemoms