openpnp/openpnp #720: Footprint gets square with 100% Roundness

kakemoms, 2018-04-18:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Footprint shows up as square in the fiducial pipeline, even when specified as 100% round.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Add a package and part for a fiductial.
2. Reset fiducial pipeline to default (machine default first, then go reset part fiducial pipeline).
3. Add a footprint to the existing package & part. First specify it as 1mm wide, 1mm length. Then edit the roundness to 100%.
4. Click edit pipeline on part fiducial. It shows up as square.

## Expected Result
Expected round fiducial.

## Actual Result
Got square fiducial

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
By adding a new package->add footprint, then edit roundness before specifying width and length, this will give a round fiducial. So it seems like this bug is connected to adding a footprint and not specifying roundness before width and length.

vonnieda, 2018-07-27:
@kakemoms I wasn't able to reproduce this. I followed your instructions step by step and got the expected round fiducial. Can you verify and reopen if this is still happening?

closed 2018-07-27 by vonnieda

markmaker, 2019-04-18:
Hi @vonnieda 

I just stumbled on the same. The first shot is from the Parts browser / ReferenceFiducialLocator tab / Edit button. The second is from the global Vision / Fiducial Locator / ReferenceFiducialLocator tab / Edit button
[image]

I looked into the code and despite all efforts I was completely unable to find the reason. Even my breakpoints weren't hit. One thing is funny:

https://github.com/openpnp/openpnp/blob/5131a7483863394f55ce70249cff3a145bfed80e/src/main/java/org/openpnp/model/Pad.java#L200-L203

vs.

https://github.com/openpnp/openpnp/blob/5131a7483863394f55ce70249cff3a145bfed80e/src/main/java/org/openpnp/model/Footprint.java#L199-L206

But I'm afraid I don't understand why those two methods exist or how they are related. I tried to change both  to `/ 100.0` which seems logical given the overall handling of roundness but it did not change a thing. It almost seems as if the footprint data in the XML has a life of its own. 

I propose reopening this.

_Mark

markmaker, 2019-04-18:
OK sometimes it helps explaining it. Found it out: the body width/height adds a shape too! 

https://github.com/openpnp/openpnp/blob/5131a7483863394f55ce70249cff3a145bfed80e/src/main/java/org/openpnp/model/Footprint.java#L49-L62

So we need to set body width/height to zero for most packages i.e. it is not a kind of "overall dimension" of the footprint. A misunderstanding what "body" means on my part. 

What is the correct usage of that? Use it for the thermal pad? I't can't really be "the body" as no-leads packages have the pads **underneath** the body, so it would again blot them out. 

No need to reopen the issue but perhaps another label would help people like me. :-)

_Mark

vonnieda, 2019-04-21:
@markmaker Body width and height were originally intended to be used by bottom vision to provide a way for the algorithm to check for missing parts. I think I'd still like to use it that way, but I never got around to implementing it, and as you saw it affects the way the template is generated for fiducials, which is not really what was intended.

For what it's worth, the Fiducial Wiki does mention to set those values to 0: https://github.com/openpnp/openpnp/wiki/Fiducials#creating-a-fiducial-package