openpnp/openpnp #721: Bottom Vision Safe Z problem

olinek2, 2018-04-20:
# Problem Report
## Describe the Issue
In some cases Bottom Vision is making movement without Safe Z. After picking up component it is making XYZ staight movement to focal point of camera without XY and then Z movement. It can make collision with some pick and place parts, like feeders if camera focal point is much lower than rest of machine.

## Steps to Reproduce
1. Start OpenPnP and turn on.
2. Click the Power On button.
3. Home machine.
4. Pick up part manually by button (Parts Tab) from some feeder (part should have bottom vision option disabled).
3. Set "Enabled?" option for vision at this particular part without clicking apply button.
4. Click "Test Alignment", ang XYZ movement without Safe Z will occur. After clicking to Apply option machine will make Safe Z movement.

## Expected Result
Machine should move in XY to camera focal point and then with Safe Z option made movement in Z axis.

## Actual Result
Machine is making straight XYZ movement to focal point of the camera without using Safe Z option and it can end with collision to another parts of PnP.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Screenshot from log file when it had happened.
[image]

vonnieda, 2019-02-16:
I am not able to reproduce this. If it's still happening, can you please send your machine.xml so I can try it?

vonnieda, 2019-02-22:
Closing due to unable to reproduce and no follow up. Please reopen if you are still experiencing this and can supply more information.

closed 2019-02-22 by vonnieda