openpnp/openpnp #723: OS freezes when starting "Edit Pipeline" in feeder settings

olablt, 2018-04-25:
# Problem Report
## Describe the Issue
I configured Gcode driver successfully, added USB camera to my cameras list and can not setup the vision for the openpnp. I am using latest Fedora linux.

## Steps to Reproduce
Going through steps in https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration:-Feeders

1. Start OpenPnP
2. Click Feeders tab
3. Create new feeder by pressing "+"
4. Press "Edit Pipeline"

## Expected Result
I expect to be able to edit Pipeline

## Actual Result
I see Pipeline window but system is freezed - only mouse pointer moving but I am unable to click anything

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Commind line log:
[image]

olablt, 2018-04-28:
any ideas how I can debug the problem? maybe some dependencies are missing for vision to work?

markmaker, 2018-04-28:
Do you have Java 8? When I had Java 9 I couldn't edit pipeline (though that was on Windows). Maybe it is related? 

See here:
https://groups.google.com/d/msg/openpnp/fs9gefGd15U/ZB38U3o-AQAJ

olablt, 2018-04-29:
Thank you @markmaker for suggestion.

I am on the latest Fedora with the Java 8

[image]

olablt, 2018-04-29:
well.. it is working after I changed the default java from openjdk to downloaded install 

```
sudo alternatives --config java
```

closed 2018-04-29 by olablt