openpnp/openpnp #724: Non-zero Y home does not work with non-squareness factor

markmaker, 2018-04-25:
# Problem Report
## Describe the Issue
When having a non-zero Y home, the non-squareness factor X offset is not applied from Y home but from Y0.

## Steps to Reproduce
Can be reproduced as follows:
1. Configure machine.xml Y axis with non-zero home-coordinate.
2. Configure non-squareness factor non-zero.
3. Home the machine (best observed with visual homing fiducial).
4. Jog 1mm to right then 1mm left. 
5. Machine will be off in X by non-squareness factor * Y home. 

## Expected Result
Non-squareness factor must be considered when homing. 

## Actual Result
Non-squareness factor is ignored, machine is off.

## Possible solutions
1. Clean solution: calculate non-squareness factor X offset with crossing point at Y home. Caveat: update of OpenPNP will slightly shift the X coordinate system for existing machine setups with Y home != 0.
2. Send POST_VISION_HOME_COMMAND with coordinates transformed with non-squareness factor. Caveat(?): does only work with visual homing (then again, it might only really be relevant with visual homing).

## Proposition
Change [GcodeDriver.java:513](https://github.com/openpnp/openpnp/blob/998241d4942309d664721bf5497bdd77f7f7a0f4/src/main/java/org/openpnp/machine/reference/driver/GcodeDriver.java#L513)
```
command = substituteVariable(command, "X", x + nonSquarenessFactor * y);
command = substituteVariable(command, "BacklashOffsetX", x + backlashOffsetX + nonSquarenessFactor * y); // Backlash Compensation

```
into :
```
command = substituteVariable(command, "X", x + nonSquarenessFactor * (y - yAxis.getHomeCoordinate()));
command = substituteVariable(command, "BacklashOffsetX", x + backlashOffsetX + nonSquarenessFactor * (y - yAxis.getHomeCoordinate())); // Backlash Compensation
```

cri-s, 2018-04-25:
Just as note, Homing coordinates are not Location 0 coordinates.
As example location 0 is front left edge and homing switch are back right edge, then the homing, depending
on controller need to be the offset to Location 0, negative or positive in order that Location 0 is correct.

Further HomeCoordinate could be big negative number and at least in past there was some problems
if location 0 is not 0,  if global coordinates changes the sign.

markmaker, 2018-04-26:
Yes that is exactly the case with my machine. I'm homing to the max Y endswitch, which is at ~607mm as seen from the min Y endswitch, I'm retracting 7mm to make it a round number then set this to my Y home at 600mm. With a non-squareness-factor of -0.001100 after visual homing - and jogging 1mm to the right, then 1mm to the left, I'm off -0.66mm in X from the homing-fiducial.

cri-s, 2018-04-26:
The purpose of homing is to have repeatable positions.
You assumtion is wrong. you cannot have before and after homing the
same coordinates.
before visually homing, the squaring calculation is not used. After
homing it get used.
There are two exceptions, if coordinate is 0, then it's the same.
In you'r case, you should not set homing y to 600, but to 0 and on y
you have negative
coordinate system. It is not important that zero coordinate is on left
front edge,instead
it is important to follow the right hand coordinate rules. If you
still want having 0 on
left front edge, then you should use G54 and G10 L20 as explained in
every CNC book.

markmaker, 2018-04-26:
>In you'r case, you should not set homing y to 600, but to 0 and on y you have negative coordinate system. 
I think OpenPNP clearly _does intend to support_ non-zero home coordinates, otherwise it would not have a home-coordinate attribute in the <axis/> element of machine.xml and variables and methods in its machine model. 

With visual homing, [this is also documented and through the variables clearly supported behavior](https://github.com/openpnp/openpnp/wiki/GcodeDriver:-Command-Reference#post_vision_home_command). If it were not supported, the variables would always be zero and therefore useless. The authors would not have gone to the trouble to provide these variables, if non-zero home coordinates were not intended.

> it is important to follow the right hand coordinate rules
Of course! But adding offsets i.e. by homing to non-zero coordinates does never change that. 

>If you still want having 0 on left front edge, then you should use G54 and G10 L20 as explained in every CNC book.
You are probably right, that the non-standard Raprap / [Smoothieware G28](http://smoothieware.org/g28) that I am using is actually a $H + G54 rolled into one. However please realize that using G54 after having homed to X0 Y0 _changes nothing_. The machine position would still be off by non-squareness-factor*Y, because the controller does not know about it.

cri-s, 2018-04-26:
The visual homing coordinates could be different to machine homing coordinates.
Immagine the trouble, if visual homing is near the board. It must be
at board height,
and the board align structure is usually a good candidate for that.
As example machine coordinates goes from 0 to 600 and homing value is at 200.
Now with you'r proposing depending if feeder is less then 100 or more
then 100mm on Y,
the squareness offset change direction.
Actually in this case,  Y=100 , X+= -0.11mm,  Y=300 , X+=-0.33mm .
You proposal now,     Y=100 , X+= +0.11mm,  Y=300 , X+=-0.11mm

Resulting on 0.22mm difference just on 100mm delta in both directions
relative to the
required value.
Some machine use position near Camera, but have the same issue.

cri-s, 2018-04-26:
Basically, after homing dot is set to 0, you should travel with
machine, not use external ruler to 0 position,
read the displayed machine position. This require not having setup
already nozzle offsets,
or if you have set it up, add annother temporary nozzle with offset 0,
and then use
the displayed value as homing coordinates. Home and update the homing
dot location.
The setup is basically simple, but require determinate steps, or if it
is made out of order,
usually require checking the logs and doing some math or creating a
temporary nozzle
in order to get access to real values.

2018-04-26 15:25 GMT+02:00, Cri S <phone.cri@gmail.com>:

markmaker, 2018-04-26:
I'm afraid I can't follow your last two posts. However I have since changed my proposition, see the original post. Maybe it becomes clearer what the problem actually is. Thank you for this discussion as it made me think this through more thoroughly, resulting in the changed proposition. 

However I'm still convinced homing to max/to non-zero is OK and should and with the proposition _can be_ supported by OpenPNP.

markmaker, 2018-04-26:
I disagree. 

You first said one should always home to 0, 0 and accept that Y axis may be (partially) negative. OK, let's consider that. 

If you look at my new proposal, you see that the actual compensation is 
`xHomeCoordinate -= nonSquarenessFactor*yHomeCoordinate;`

So in your proposed case this does **nothing** as yHomeCoordinate = 0. If you follow the code carefully you'll see that this will result in exactly the same outcome as before.

Now for the non-trivial non-zero Y home case:

Regarding "sign inversion" I just tested this modification with my machine. The sign is correct. Now everything works as before, except the coordinates displayed after homing are correctly reflecting the **inversed** non-squareness compensation (see screenshot):
[image]

When I jog 1mm right, then left, I'm perfectly back over the fiducial. I captured the location immediately after homing as the park location and can now always return precisely over the homing fiducial (not so before!!).

NOTE that the compensation has to be **inversed**, because - as you pointed out correctly - we are still in (non-square) machine coordinates until homing is complete. Only after homing is complete, we switch over into the OpenPNP (square!) coordinate system. BUT: in order for this switch to be seamless, we need to inverse-compensate for the non-squareness factor **for the current Y machine position**. That's the only difference there. 

NOTE that in the current code after the first move command, the internal coordinate is obviously correct again, as the controller is working in absolute mode and will just do as told. The bug just shows when you move back to the home coordinates and are surprised to be off from the fiducial. 

I checked all the other coordinates captured before this modification and it behaves exactly as before. Including the 6-nozzle tip holder where precision is crucial and strip feeder sprocket holes at very different X and Y coordinates. 

> hat is not the case in you'r setup as you use min or max value and this don't produce sign inversion and biased offsets, but it is only a special case and a lot of pnp need to have vision homing not at min or max Y location.

I can't follow this argument. The non-squareness transformation we're talking about is a [simple 2D linear transformation](https://en.wikipedia.org/wiki/Transformation_matrix#Shearing). As I have both a non-zero Y home and a non-zero non-squareness factor, my test case is non-trivial and therefore complete. If two different points match (as I tested and confirmed), it will always match, for any home Y and any non-squareness factor (including the trivial zero cases). That's just how Linear Algebra works (and why it is so mighty).

cri-s, 2018-04-26:
There are machines, that need to have the calibration fiducial near board0
or up looking camera, because of the implicit requirement of having the
fiducial near board height.
That setup need to be homed with final homing dot and squareness corrected
fiducial location . as example if the homing dot is at 100:200mm this is
the homing coordinate and the fid-fot location could be -500:300 as
example. Here you have the sign inversion for feeders and maybe inside
board. With min or max y position this problem is not oresent, but not all
can do that,

markmaker, 2018-04-26:
Yes I understand but my proposition does not change anything for that case _or any case_. Like I tried to explain, the controller is homed to exactly to the same G code (non-square) coordinate as with the present code. All moves subsequently go to exactly the same coordinates as before. The only difference is in the moment immediately after homing, when OpenPNP internally assumed to be at the wrong (square) coordinates (off by the non-sqareness compensation). I don't know how to explain differently.

markmaker, 2018-04-26:
> Here you have the sign inversion for feeders and maybe inside board. With min or max y position this problem is not oresent, but not all can do that,

Perhaps I should point out that the signs cancel each other out. As Y changes sign into the negative, so does the X offset from the non-squareness factor. Like I said it's a simple linear transformation, meaning there can't be any jumps or corners. The term "linear" comes from "along a line". To illustrate, I used the [documentation image](https://github.com/openpnp/openpnp/wiki/GcodeDriver#non-squareness-compensation):
[image]

So the two coordinate systems meet where Y is zero. But they extend in perfectly continuous fashion from there.  

It does **NOT** work like this:

[image]

There is no sign reversal.

cri-s, 2018-04-26:
If you have a pnp like this, then the only possibility for homing dot
is the up/downlooking
camera with it's illumination. The rail in front is fix, but the
camera is unable to reach it,
and the back rail is movable, not suitable for having a homing dot.
Having a homing dot means less meccanical service intervals and less problems.
The camera and stopper is movable, and on startup, the downlooking
camera need to
detect the position of uplooking camera.

> Perhaps I should point out that the signs cancel each other out. As Y
> changes sign into the negative, so does the X offset from the non-squareness
> factor. Like I said it's a simple linear transformation, meaning there can't
> be any jumps or corners.
No, there is no sign inversion at homing dot for X and Y coordinates,
but if you subtract the homing dot there are.
Again, if the homing coordinate is 200, with g92 the axis is set to
200 and the 0 position,
where the axis inverts, are 200mm away, but you proposal calc with
that position.
It could be a bit confusing, that the homing position is in reality
the homing dot offset
to real zero for different setups,.

https://github.com/openpnp/openpnp/wiki/GcodeDriver#non-squareness-compensation
Yes, here the minimum coordinate are used , 0:0 , you use the maximum for Y, but
if You use values between, for complexer situations, it works as it
should, not with you'r
modification.

The documentation is correct, explaining the simplest usage. There
exist other user case.

markmaker, 2018-04-28:
> Yes, here the minimum coordinate are used , 0:0 , you use the maximum for Y, but if You use values between, for complexer situations, it works as it should, not with you'r modification.

Things are **not** as complicated as you describe them. I try to explain why. 

There are two coordinate systems.

1. The first coordinate system is the G Code controller's machine coordinate system. Due to imperfections in the simple mechanics and tools we mostly use (not to mention varying skills), this machine coordinate may be physically non-square. There could be many imperfections like bent rails, but the only one OpenPNP can compensate for, is where the angle between the X and Y axes is not perfectly 90° (typically by far the largest imperfection in a maker built machine, I guess). 
2. The second coordinate system is the internal OpenPNP coordinate system that must be square, because we want to blindly populate the PCB which is professionally manufactured and therefore pretty square. 

Whenever a square coordinate from OpenPNP is sent to the machine, the non-squareness factor is [applied in GcodeDriver.java:513](https://github.com/openpnp/openpnp/blob/998241d4942309d664721bf5497bdd77f7f7a0f4/src/main/java/org/openpnp/machine/reference/driver/GcodeDriver.java#L513)
```
command = substituteVariable(command, "X", x + nonSquarenessFactor * y);
command = substituteVariable(command, "BacklashOffsetX", x + backlashOffsetX + nonSquarenessFactor * y); // Backlash Compensation
```

As you see, this is very simple: `x + nonSquarenessFactor * y`. There is no `if`, no `abs()` handling signs, just a linear expression. This means that this transformation works seamlessly from the negative to zero and into the positive for all numbers x, y and nonSquarenessFactor. Mathematically it is absolutely irrelevant whether your homing location, fiducial or your origin is at the maximum, the minimum, in the middle or even beyond endstops of any axis. 

On moves, the internal coordinate is transformed into the machine coordinate. The information flows from OpenPNP to the machine. There is one tiny moment, when this goes the other way around. Just after electromechanical and/or visual homing, you pin down the machine coordinates with G92. There you have to realize that the information flows the other way. The information **is the current machine position** and it flows back **into** the current internal position of OpenPNP. Therefore we have to take the coordinate from the machine coordinate system and transform it into the internal OpenPNP coordinate system. If Y home is zero, this can be ignored, because with `x + nonSquarenessFactor * y` there is no difference (multiplication by zero). But if Y is non-zero, we must compensate for that. Because the information is flowing the other way, we must inverse the non-squareness transformation. This is easy: we just flip the sign from plus to minus: `x - nonSquarenessFactor * y`.  It is easy to see, that the inverse of adding something is subtracting that same something, right? 

Again this is very simple.  Again no `if`, no `abs()` handling signs just a linear expression. Again this means that this transformation works seamlessly from the negative to zero and into the positive. Again there can be no difference whether your homing location, fiducial or your origin is at the maximum, the minimum, in the middle or even beyond endstops of any axis. 

Lets play this through: Say we home to X0 Y500 and we have a non-squareness factor of 0.001. So for the machine the home is at X0 Y500. In the current, faulty code, OpenPNP now **also** thinks it is at X0 Y500. But when OpenPNP wants to move to X0 Y500, it uses the non-squareness transformation `x + nonSquarenessFactor * y` and sends G0 X0.5 Y500 to the machine. If you homed using a fiducial you will now notice the offset/bug. 

Now with my bugfix OpenPNP will transform the machine homing location X0 X500 back using `x - nonSquarenessFactor * y` and correctly think it is at X-0.5 Y500 after homing. When it now wants to move back to the homing location at X-0.5 Y500 it uses the non-squareness transformation `x + nonSquarenessFactor * y` and sends G0 X0 Y500 to the machine. Perfect!

You can think this through with any signs and numbers. If you find a case where this does not work, please document! You'll probably get the Nobel price for revolutionizing Linear Algebra :)

cri-s, 2018-04-28:
Here you describe the existing system, you propose a change
subtracting the homing (dot)
location from actual Y coordinate as i cite:
"command = substituteVariable(command, "X", x + nonSquarenessFactor *
(y - yAxis.getHomeCoordinate()));"

and if you retrofit pnp that have limitation  where the camera moves
and where the pcb heigth is 70 to 180mm over height of  the solid
plate the options are
limited for placing homing dots and without homing dot it bumps to the
structure requiring
frequenter meccanical mainteainence and without the possibility of
verify or correct the
accuracy of homing. This machines usually further don't have camera
coverage for
pluggable feeder, only for the tray/strip feeder position. Depending
on pcb dimension,
the tray or strip feeder shelf cannot be loaded.

markmaker, 2018-05-01:
Merged. 
https://github.com/openpnp/openpnp/pull/725

closed 2018-05-01 by markmaker