openpnp/openpnp #730: EOL at EOF in xml files

ticso24, 2018-05-14:
# Read This First

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
OpenPnP saves the  xml files without line termination at end of file.
The issue is that I often edit them with vi, which adds an EOL to the end and this creates additional difference, when I manage the files in git.

## Steps to Reproduce
## Expected Result

## Actual Result

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
I like to have OpenPnP to write the files with usual end of line termination.
It is not a big thing, but somehow bugs me.

## How Will It Look?

## Why Do We Need It?
It is nothing we really need and it maybe is just a personal thing, but I don't think the change would hurt anyone else and maybe I'm not alone.

## Provide Examples

vonnieda, 2018-05-15:
Agreed - would be a good fix. Might end up being surprisingly hard, since we use a third party library for this - but certainly worth looking in to.

pfried, 2018-08-01:
Only for reference, POSIX standard should have a newline at the end of a file:
https://stackoverflow.com/questions/729692/why-should-text-files-end-with-a-newline

See https://stackoverflow.com/questions/1050640/vim-disable-automatic-newline-at-end-of-file for an alternative "solution" of changing vi/vim settings.

The encoding should be fairly easy since you could just write before closing. The bigger question is if it confuses the decoder which I guess doesnt.

vonnieda, 2018-08-02:
Fixed in above commit.

commit cb39077e (2018-08-02): Fixes #730 by writing an additional newline at the end of machine.xml, parts.xml, and packages.xml.
commit 2f48b1b7 (2018-08-05): Merge branch 'develop' into feature/guify-signalers

* develop:
  Changes for opencv upgrade
  Updates required to move to opencv-3.4.2
  Fixes #730 by writing an additional newline at the end of machine.xml, parts.xml, and packages.xml.
commit 60acea64 (2018-09-27): Merge branch 'develop' into feature/i18n

* develop:
  Adds a OffsetTransform to GcodeDriver that can be used to trick OpenPnP into working with Shared C machines. See https://groups.google.com/d/msgid/openpnp/CA%2BKNHNxW4vE_rvi7MSLvnhsiw4BAE0xyJN%2BeEP%3D5dq5x45WHEA%40mail.gmail.com?utm_medium=email&utm_source=footer for details.
  updates changes for https://github.com/openpnp/openpnp/pull/765
  Adding a nozzle offset wizard, solves #307
  Clear search field on new feeder created, fixes #714 Correct wrong javadoc param
  Add a comments field to the placements table. Fix a bad property name f [... cut]
closed 2018-08-02 by vonnieda