openpnp/openpnp #732: Cannot open Named CSV files

DaneMcGowan, 2018-05-16:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
The "Import Named CSV" option always returns the error "Unable to parse CSV File Names". I have followed the file format as precisely as I can. I even copied the example from https://github.com/openpnp/openpnp/wiki/Importing-Centroid-Data#others directly into a CSV file.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Copy the example CSV data into Notepad++, save as a .csv.
2. Start OpenPnP
3. Click File -> Import Board -> Named CSV.
4. Browse to the example file
5. Error will display

## Expected Result
I expected the parts list to appear.

## Actual Result
I get the error described.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
I am attaching the file I am trying to load as test.txt. I changed the extension to placate GitHub. To reproduce the error, the extension should be changed back to .csv.
[test.txt](https://github.com/openpnp/openpnp/files/2010575/test.txt)

-----------------------------------------------------------------------

vonnieda, 2018-05-17:
Your editor appears to be inserting a Unicode BOM at the beginning of the file. Make sure you tell your editor to save the file as plain ASCII text or UTF-8. I've attached the same file with the BOM removed, and it imports correctly.

[fixed.txt](https://github.com/openpnp/openpnp/files/2011274/fixed.txt)

closed 2018-05-17 by vonnieda

DaneMcGowan, 2018-05-17:
Thanks for a the quick response, super helpful!