openpnp/openpnp #733: How to attach camera jogging to a camera?

DaneMcGowan, 2018-05-16:
# Problem Report

Camera Jogging is described here;

Camera Jogging
Now that the camera is configured, click and drag anywhere in the camera view. You'll see a white line follow the mouse cursor. When you release the mouse button the camera should move to the location you were pointing at. This is called Camera Jogging and is the easiest way to move the machine around when setting things up. You can also hold Shift and then Left Click within the camera view to do the same thing.

This does not work for me. The white line appears and otherwise seemingly-functional machine does not move. Oddly, if I drag around the simulated camera that appears during the default set up, the camera DOES move. Is there a way to attach jogging to a specific camera/device?

vonnieda, 2018-05-17:
Check the following:
1. Make sure you've deleted the default simulated camera. OpenPnP uses the default camera for the head.
2. Make sure the camera is assigned to the head, and is set to Looking: Down.
3. Make sure units per pixel have been set.

If none of that works, please post your [machine.xml](https://github.com/openpnp/openpnp/wiki/FAQ#where-are-configuration-and-log-files-located).

DaneMcGowan, 2018-05-17:
Thanks again for the quick reply. I re-calibrated the units-per-pixel and it has been working fine since. I must have taken a mist-step along the way the first time.

closed 2018-05-17 by DaneMcGowan