openpnp/openpnp #734: recent spi change for Partalignment

cri-s, 2018-05-21:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
PartAlignment don't necessary give out angle information and the printf is unnecessary.
Please move it into the bottomVision PartAligned function and not in spi.
As example the gripper aligners, cyberoptics lasers or camera systems having the same principe
(luna is a example)  are all examples that run and are not able to give out ony useful debug lines.

## Steps to Reproduce
I can share the generic Partalignment code and scripts, but the problem using this Aligners
is that the machine must have it's own Part orientation, because the sequence of aligning 
is essential. 

## Expected Result
remove the specific code for bottomvision debugging from pnpJob code and revert back the
last spi interface.

## Actual Result
Actually when compiling a dummy toString into the interface, the log file is empty of useless
lines, not nice to have when having over 6 nozzles.

vonnieda, 2018-05-22:
Hi @cri-s,

The SPI change was made to an inner class, not the interface: https://github.com/machinekoder/openpnp/blob/4382df424b793c4577b9a1ca86d0c00008be4595/src/main/java/org/openpnp/spi/PartAlignment.java#L38

The inner class returns a Location, which includes rotation information. I don't see the harm in showing this information, even if the result is 0,0,0,0. 

I don't see how this can cause any problems, since it's simple a toString (which can be overridden) and a few debug statements. If this is causing a problem in your code, please post the code. 

Thanks,
Jason

Link: https://github.com/openpnp/openpnp/pull/731

cri-s, 2018-05-22:
> Hi @cri-s <https://github.com/cri-s>,
Hi @vonnieda

> The SPI change was made to an inner class, not the interface:
> https://github.com/machinekoder/openpnp/blob/
> 4382df424b793c4577b9a1ca86d0c00008be4595/src/main/java/org/
> openpnp/spi/PartAlignment.java#L38
 Nice justification.

> The inner class returns a Location, which includes rotation information. I
> don't see the harm in showing this information, even if the result is
> 0,0,0,0.
Compiling the code give error because not implemented function. Adding this
function gives null pointer assertion when running the code.

> I don't see how this can cause any problems, since it's simple a toString
> (which can be overridden) and a few debug statements. If this is causing a
> problem in your code, please post the code.

Basically the problem is the unsafe use of this code or subclass.
 As from the description of the Interface that you or someone else have
written if the alignment is hardware basRegards
 (referenced as zero in the docs)  is returned.
The last code change don't allow anymore returning zero but assumes that
always a valid "subclass" is returned.

As the purpose of this change seems to debug the output of bottom vision
alignment I have proposed to move the debug statement to that class because
in that case the angle offset  information is always available where if it
printed out inside jobprocessor that part of info is missing in case of
prerotate. Further this avoid changing the spi interface.
There are other possibility moving the debug output down in the source.

Regards Chris.

cri-s, 2018-05-22:
PS:
The error mentioned of not implemented was the result of link it with the
precompiled groovie script. Running it as script or compiling the groovie
script again resolve this, so this info was wrong. That don't change the
complaint.

vonnieda, 2018-05-22:
What specific line or lines are your concern, and what is the problem? Is
it an NPE? Please just be specific.

cri-s, 2018-05-22:
findOffsets(...) returns zero (null) as it use hardware alignment.

In reality it checks nozzle , nozzleTip and package and  decide if doing alignment or not.
Example from philips machine.
Nozzle 1 always do gripper alignment.
Nozzle 2 if packagename contains qfn or if part is less then 5x5mm and package don't contains sot does
gripper alignment, otherwise it does bottom vision.
Nozzle 3 always chain PartAlignment to next PartAlignment (BottomVision) checking the PartAlignment chain skipping the first Aligner and checking canHandle(Part).
canHandle(Part) always return true.

The problem is line 662 inside ReferencePnPJobProcessor.java , assume that findOffset
always return a subclass as it the case with Bottomvision.
Move it down to line 697 could be a fix.

cri-s, 2018-05-24:
For you'r reference, a minimal working mockup of the PartAligner used.

vonnieda, 2018-05-25:
@cri-s There does not seem to be a problem here. The code works fine, even with null. Are you using the same logging system I am? Perhaps your logger does not handle null?

```
                plannedPlacement.alignmentOffsets = VisionUtils.findPartAlignmentOffsets(
                        partAlignment,
                        part,
                        boardLocation,
                        placement.getLocation(), nozzle);

                // FORCE TO NULL
                plannedPlacement.alignmentOffsets = null;

                Logger.info("Align {} with {}", part, nozzle);
                Logger.info("Offsets {}", plannedPlacement.alignmentOffsets);
```

```
INFO: Align id R0805-1K, name null, heightUnits Millimeters, height 1.000000, packageId (R0805) with N1 N1
2018-05-24 20:39:46 [main] org.openpnp.machine.reference.ReferencePnpJobProcessor.doAlign()
INFO: Offsets null
Placing R0805-1K for R1.
```

The logging system handles the null correctly. There is no error.

cri-s, 2018-05-25:
You are correct. I use the same logging system, but use realtime java with realtime GC, that precompile/transform the class file and this don't like the null. I should have checked it before.

closed 2018-05-25 by cri-s