openpnp/openpnp #737: Wizard problems multiple sheet.

cri-s, 2018-06-12:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
I need a feeder with multiple Wizard pages, like Gcode driver.
Unfortunatly if i just have a feeder that overrides getPropertySheets() it don't work.

public class StripesFeeder extends StripFeeder {
    @Element
    private Feeder fdr = new StripFeeder();

....

    @Override
    public PropertySheet[] getPropertySheets() {
        return new PropertySheet[] {
                new PropertySheetWizardAdapter(fdr.getConfigurationWizard(), "Stripe")
               ,new PropertySheetWizardAdapter(    getConfigurationWizard(), "Stripes")
        };
    }

}

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:c
copy TubeFeeder and add this function. 

    @Override
    public PropertySheet[] getPropertySheets() {
        return new PropertySheet[] {
                new PropertySheetWizardAdapter(   .getConfigurationWizard(), "Stripe")
               ,new PropertySheetWizardAdapter(    getConfigurationWizard(), "Stripes")
        };
    }

## Expected Result
What did you expect to happen? Be descriptive.
Wizard with two tabs , instead of only one configuration tab.

## Actual Result
The code from AbstractFeeder is executed (apparently) displaying only one wizard tab.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

vonnieda, 2018-06-14:
Yes, currently the Feeders tab does not use the newer PropertySheets system, it uses the older getConfigurationWizard() system which returns the wizard defined by the AbstractFeeder. As you noticed, the Machine Setup panel works correctly.

I think it should be an easy conversion to switch the Feeders panel to use the same code as the Machine Setup panel. There may be a few feeder wizards that have to be modified. I think some of them merge in parts of the ReferenceFeederConfigurationWizard. We'd just need to spot each one and make sure it appears the same before and after the change.

vonnieda, 2018-06-14:
One note: As a simple workaround you can have your wizard return a JTabbedPane and build your tabs internally.

vonnieda, 2018-06-14:
I went ahead and made this change in develop. Give it a try and see what you think.

commit ae08caac (2019-05-19): Merge branch 'develop'

* develop:
  Changes the Feeders tab to use the property sheets system, allowing a feeder to have multiple configuration panels. Fixes #737.
  reintroduce multiple uplooking camera support
  ReferencePnpJobProcessor: print offset in debug output
  ReferenceBottomVision: add nicer debug output
  The internal X coordinate set after homing is not compensated for non-squareness after homing to non-zero Y. The non-squared machine X is used directly. This results in incorrect coordinates displayed immediately after homing. When capturing these coordinates and returning to the [... cut]
closed 2018-06-14 by vonnieda