openpnp/openpnp #739: Save machine.xml during a Session

KarlZeilhofer, 2018-07-02:
# Feature Request
## Describe the Feature
Currently the machine.xml gets written to disk only when OpenPnP gets closed properly. It would be great to have an action, so the machine.xml is saved immediately. 

## How Will It Look?
A menu entry in Menu -> Machine -> Save machine.xml

## Why Do We Need It?
1. Configuring pipelines and feeders and configuring the machine can be very time intensive.
2. A crash of OpenPnP still happens from time to time and changes get lost. 
3. Sometimes OpenPnP runs for days, and all the changes could get lost
4. I think this is useful for every one.

sp-apertus, 2018-07-02:
+1 on that as I often just close openpnp after setting up a job to make sure I don't loose all my changes/work to an unexpected crash/hang.

vonnieda, 2018-07-02:
Sounds good to me, but let’s call it Save Config since we’ll also need to
save parts and packages.

Jason

vonnieda, 2018-07-03:
Added the menu command.

closed 2018-07-03 by vonnieda

KarlZeilhofer, 2018-07-03:
@vonnieda I've changed the title and description to be applied for the 3 config files.

vonnieda, 2018-07-03:
Thanks @KarlZeilhofer - in case you missed it in the message above, I added this feature to the develop branch last night at https://github.com/openpnp/openpnp/commit/94b2216263a4440aead38fce3f7988b1a94b61ac

commit 94b22162 (2018-07-03): Adds a Save Config menu command under Machine. Fixes #739