openpnp/openpnp #754: Retrofitting a Zeva 460 Part 1 - documentation files/manual is missing

anubisis, 2018-07-21:
Hello,

I have problem to download Zevatech PM460 manual files from the section "Retrofitting a Zeva 460 Part 1". On the Yahoo groups, there is problem with one file but no one is reacting almost for year for that issue. 

I don't know where else should I ask, than in openpnp (because the PM460 is still in use).
Please, can somebody, who have this complete documentation of the PM-460, upload it and add direct link to the "Retrofitting a Zeva 460 Part 1" page.
I have one this machine and I want it to rebuild, but without documentation it goes very slowly.

Thanks in advance and small apology, because this is not issue related with openpnp.

vonnieda, 2018-07-21:
@dandumit Could you possibly help with this?

dandumit, 2018-07-21:
Hello ,
I will have a look.
Thank you,
DAniel