openpnp/openpnp #760: Camera Performance

vonnieda, 2018-08-02:
Some notes on improving camera performance, and performance in general:

Test setup is Facetime camera added as OpenPnPCapture machine camera at 1280x720, 29 FPS, 2vuy.

* No cameras enabled, barely any CPU at all. The cameras are the bulk.
* Using Facetime camera, with the camera shown, about 40% CPU, Show None about 20% CPU.
* Maybe get rid of the whole continuousCapture system entirely. Only thing that uses it is CameraView and that could just run it's own thread. Would vastly simplify all the camera code.
* Commenting out broadcastCapture shows that broadcast/render is about 20% CPU, and the capture itself is about 20%.
* internalCapture runs transformImage, but then the broadcaster potentially drops the frame if it's too fast for the camera FPS, meaning all that work was wasted.

mgrl, 2019-01-10:
I found this a good place to add some screenshots from the new adaptive settle feature - just to throw in some numbers:

I made three sets, one with the camera at 30fps, one at 60fps and one at 120fps.
The logs are for a fid check repeated 4 times on my machine. the following 3 sets were taken with a settle time of -200. the produces images are all perfect (at least that I see, I have no CV eyes ;)

### 30fps
[image]
[image]

### 60fps
[image]
[image]

### 120fps
[image]
[image]

So it seems the higher the fps the faster the processing... but is it due to reduced resolution and processing overhead or due to faster image streaming?... Anyway, the found autosettle-times are far higher than theoretically possible.

Here additionally two images from the pipeline for a plus 120 settletime (so conventional):
[image]
[image]
please do download them and then switch back and forth. you will see, that the picture is invalid, since it shows parts moving only partially, not on the whole image. so I would need a higher settle time for sure. the new feature allows me to use adaptive settle times and saving time...

mgrl, 2019-01-10:
and to compare additionally one set for raw camera mode:
### 30fps in yuy2 mode
[image]
[image]

you can see quite often score of 0, what means the processor got 2 equal frames from the camera. this leads me to the assumption, that raw streaming @30pfs tunnels to openpnp faster than with mpg even @120fps?

markmaker, 2019-01-10:
Hi mgrl

thanks, that's very interesting! 

Do you think it's possible that JPEG compression adds _artificial_ differences? 

After all the compression is optimized for the human eye, not CV. And if you think about how it works, I can imagine how due to the DCT there could be large pixel differences on sharp edges that are simply compression artefacts (ringing), caused by even the slightest weight disturbance in the source.
https://en.wikipedia.org/wiki/JPEG#Quantization

Note that MJPEG works intraframe-only, so temporal changes are in no way suppressed:
https://en.wikipedia.org/wiki/Motion_JPEG#Encoding

When I programmed a [camera trap](https://en.wikipedia.org/wiki/Camera_trap)  years ago, I had to compare the viewfinder lowres JPEG pixels not only to its predecessor, but also to the 8 surrounding pixels. I took the minimum of the 9 differences as the pixel difference. It worked quite well and also ignored creeping shadows due to the sun advancing.  

_Mark

felixekman, 2019-04-25:
I stumbled upon a camera (Leap Motion) that might be interesting for OpenPnP uses as it has a global shutter and UVC (Universal Video Class) interface. They also recently opened up their firmware and has some great OpenCV applications. Maybe this can be a fast, high-performance bottom vision camera with built-in LEDs and in a slim package for us?

> "This gives you low-level controls such as LED brightness, gamma, exposure, gain, resolution...."

Would love to hear peoples thoughts on this. I've managed to snap one up on eBay for cheap and will do some additional research myself.

Some further reading:
[http://blog.leapmotion.com/leapuvc/
](http://blog.leapmotion.com/leapuvc/
)[https://github.com/leapmotion/leapuvc](https://github.com/leapmotion/leapuvc)

[image]
[image]

CircuitSetup, 2020-05-04:
I currently have my ELP up camera set to 1280x720 30FPS, MJPEG. 

If I set the Preview FPS to 30 from 10, the preview (and whatever action it is performing) will freeze more often. 

This is on a Windows 10 computer with 16gb ram, and an 8 core AMD CPU. The CPU usage will go up, but it is not maxed out (or even close - usually around 10%). Memory usage will also go up, but again, it's not maxed.

markmaker, 2020-05-04:
Hi @CircuitSetup 

setting FPS this high is currently not a good idea, The Preview will snatch these frames away from Camera Settling / Computer Vision. The same frame cannot be used for _both_ Preview _and_ Settling/Vision.

This might be the reason for the freezes.  Settling/Vision simply [starves to death](https://en.wikipedia.org/wiki/Starvation_(computer_science)). 

I have my fps at 5 and think that's plenty. 

I have asked @vonnieda  about this [here](https://groups.google.com/d/msg/openpnp/jnvon8elGzI/CFFKNwoUBQAJ), but understandably he seems to have overlooked the question at the end of that rambling post of mine.

> Out of curiosity, why are frames pushed by the captureNotifier? Why not just let the camera view invalidate itself with the chosen fps, then in its paint handler (if and when it is executed at all) pull the frame instead? But use no blocking i.e. no CaptureStream.hasNewFrame(), just use the last captured frame instead (I believe the CaptureStream does just that, If you skip the hasNewFrame()). 
> 
> If this works, it could have several advantages:
> 
> * Frames are only captured as long as a camera's view is actually visible. I believe Java Swing is highly optimized (from the olde days, when it still mattered) to only repaint what is absolutely necessary. No repaints for hidden or even just obstructed windows. Multiple repaint requests  are collapsed into one i.e. repaints also never pile up.
> * I believe users most often use just one camera view, so the other view(s) would leave the camera (and CPU) alone.
> * If the GUI is busy, nothing (bad) happens (i.e. no deadlocks, no machine captures blocked and no frames stolen, also no CPU wasted).
> * If the GUI was busy for some time (or the whole CPU struggling), no stale frames are displayed, i.e. no lag in the camera view (still stuttering, but that's half as bad).
> 
> Could this be related?
> https://github.com/openpnp/openpnp/issues/760

_Mark

CircuitSetup, 2020-05-04:
Thanks @markmaker . I don't necessarily need to set everything to 30 FPS, but the freezing is very similar to what I'm seeing with it set at just 10 FPS. I'll change it to 5 and see how everything performs.

markmaker, 2022-02-03:
I'm closing this for the same reasons I described in #392.

closed 2022-02-03 by markmaker