openpnp/openpnp #764: Can not run 2 webcams in parallel using single USB-port while using OpenPnpCaptureCamera.

ldpgh, 2018-09-08:
## Describe the Issue
Can not run 2 webcams in parallel using single USB-port while using OpenPnpCaptureCamera. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP with empty .openpnp
2.1 Add 2 cameras (same type) ... one down-looking at the head, 2nd camera up-lookings as regular camera ... see screenshot. ATTENTION The attached screenshot runs both cameras on two separate USB-ports of the computer. Screenshot shows that the webcam base-setup seems to work.
2.2 Camera = https://www.trust.com/de/product/16429-spotlight-webcam 

## Expected Result
If I understand right, both webcam can operate in parallel when using OpenPnpCaptureCamera on a single USB-port. I'm aware of the difficulties associated with this setup. Maybe got just wrong impression.

## Actual Result
Just one camera provides the video stream. The second camera returns black screen. The attached (functioning) screenshot has the 2nd camera attached to a different USB-port of the computer.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include URL ...
https://pastebin.com/uft10PpV

ldpgh, 2018-09-08:
The mentioned screenshot ...
https://github.com/ldpgh/rasenmaeher/blob/master/OpenPnpCaptureCamera.png

vonnieda, 2018-09-09:
@ldpgh Please click the "Format" dropdown and show a screenshot of what is listed there. Also please mention which format you had selected.

ldpgh, 2018-09-09:
Yesterdays selected format was 640x480.

Today check with 320x200 (and recreated .openpnp ... previous statement was wrong regarding fresh .openpnp) provided the same issue

Format screenshot ... https://github.com/ldpgh/rasenmaeher/blob/master/OpenPnpCaptureCamera_format_2018_09_09.png

vonnieda, 2018-09-09:
Your camera is showing only YUV formats. These are uncompressed formats and
you can’t use more than one per USB host due to bandwidth issues. You can
either use separate USB hosts or change to cameras that support MJPG
formats, which are compressed.

Jason

ldpgh, 2018-09-11:
Last comment for completeness ....
 
A mixed-setup with lowest speed setting is not sufficient to get it running trough a single USB-hub.
- Creative LIVE CAM SYNC HD 720P HD-Webcam  @    320x240,30FPS,MJPG  &  4FPS 
- Camera referenced in the 1st comment                 @    160x120,30FPS,YUY2  &   3FPS 

Setup with one camera directly connected to the    2nd (different) USB-port of the laptop   is working. But this is no surprise - just to check the cameras itself.