openpnp/openpnp #773: Exception in thread "AWT-EventQueue-0" java.lang.ExceptionInInitializerError

wholder, 2018-10-02:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
I'm trying to build OpenPnP using the latest version of IntelliJ IDEA in order to run and debug on my Mac Pro.

Info:
IntelliJ Community version 2018.2.4 (with Maven Plug in)
MacOS High Sierra 10.13.6
Java 1.8.0_144

## Steps to Reproduce
1. Import project into IntelliJ Community Edition using File>New->Project From VersionControl->Git
2. Build project.
3. Run project

## Expected Result
I expected the program to run.

## Actual Result
Code throws Exception in thread "AWT-EventQueue-0" java.lang.ExceptionInInitializerError exception.  It seems to be unable to load the translations.properties file from the "translations" Resource Bundle.  

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
2018-10-01 13:56:13 SystemLogger ERROR: Exception in thread "AWT-EventQueue-0" java.lang.ExceptionInInitializerError
2018-10-01 13:56:13 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame.<init>(MainFrame.java:979)
2018-10-01 13:56:13 SystemLogger ERROR: 	at org.openpnp.Main$1.run(Main.java:152)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(EventQueue.java:97)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:709)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:703)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2018-10-01 13:56:13 SystemLogger ERROR: Caused by: java.util.MissingResourceException: Can't find bundle for base name org.openpnp.translations, locale en_US
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.throwMissingResourceException(ResourceBundle.java:1564)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.getBundleImpl(ResourceBundle.java:1387)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.getBundle(ResourceBundle.java:773)
2018-10-01 13:56:13 SystemLogger ERROR: 	at org.openpnp.Translations.<clinit>(Translations.java:9)
2018-10-01 13:56:13 SystemLogger ERROR: 	... 16 more
2018-10-01 13:56:13 SystemLogger ERROR: Caused by: java.lang.NoClassDefFoundError: org/openpnp/translations (wrong name: org/openpnp/Translations)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.lang.ClassLoader.defineClass1(Native Method)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
2018-10-01 13:56:13 SystemLogger ERROR: 	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:335)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle$Control.newBundle(ResourceBundle.java:2640)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.loadBundle(ResourceBundle.java:1501)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.findBundle(ResourceBundle.java:1465)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.findBundle(ResourceBundle.java:1419)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.findBundle(ResourceBundle.java:1419)
2018-10-01 13:56:13 SystemLogger ERROR: 	at java.util.ResourceBundle.getBundleImpl(ResourceBundle.java:1361)
2018-10-01 13:56:13 SystemLogger ERROR: 	... 18 more

vonnieda, 2018-10-02:
@wholder Fixed by moving the translations files to the right directory. Please pull develop and try again.

closed 2018-10-02 by vonnieda

wholder, 2018-10-02:
> Closed #773 <https://github.com/openpnp/openpnp/issues/773> via 3254445

Thanks.. That got me past that confusing hurdle, but now I'm getting the
following error:

Wayne

2018-10-01 20:01:14 Configuration INFO: No packages.xml found in
configuration directory, loading defaults.
2018-10-01 20:01:26 Configuration INFO: No parts.xml found in configuration
directory, loading defaults.
2018-10-01 20:01:35 Configuration INFO: No machine.xml found in
configuration directory, loading defaults.
2018-10-01 20:01:57 SystemLogger ERROR: java.lang.Exception: Error while
reading machine.xml
(/private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib:
dlopen(/private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib,
1): Library not loaded: /usr/local/opt/ffmpeg/lib/libavcodec.58.dylib
  Referenced from:
/private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib
  Reason: image not found)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.openpnp.model.Configuration.load(Configuration.java:287)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.openpnp.gui.MainFrame.<init>(MainFrame.java:611)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.openpnp.Main$1.run(Main.java:152)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.event.InvocationEvent.dispatch$$$capture(InvocationEvent.java:311)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.event.InvocationEvent.dispatch(InvocationEvent.java)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventQueue.access$500(EventQueue.java:97)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventQueue$3.run(EventQueue.java:709)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventQueue$3.run(EventQueue.java:703)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.security.AccessController.doPrivileged(Native Method)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2018-10-01 20:01:57 SystemLogger ERROR: Caused by:
java.lang.reflect.InvocationTargetException
2018-10-01 20:01:57 SystemLogger ERROR: at
sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
2018-10-01 20:01:57 SystemLogger ERROR: at
sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
2018-10-01 20:01:57 SystemLogger ERROR: at
sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.lang.reflect.Constructor.newInstance(Constructor.java:423)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.simpleframework.xml.core.InstanceFactory.getObject(InstanceFactory.java:100)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.simpleframework.xml.core.InstanceFactory$ClassInstance.getInstance(InstanceFactory.java:233)
2018-10-01 20:01:57 SystemLogger ERROR: at
[... 58 lines cut ...]
2018-10-01 20:01:57 SystemLogger ERROR: at
java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.lang.Runtime.loadLibrary0(Runtime.java:870)
2018-10-01 20:01:57 SystemLogger ERROR: at
java.lang.System.loadLibrary(System.java:1122)
2018-10-01 20:01:57 SystemLogger ERROR: at
nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:208)
2018-10-01 20:01:57 SystemLogger ERROR: at
nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:189)
2018-10-01 20:01:57 SystemLogger ERROR: at
nu.pattern.OpenCV$SharedLoader$Holder.<clinit>(OpenCV.java:229)
2018-10-01 20:01:57 SystemLogger ERROR: at
nu.pattern.OpenCV$SharedLoader.getInstance(OpenCV.java:233)
2018-10-01 20:01:57 SystemLogger ERROR: at
nu.pattern.OpenCV.loadShared(OpenCV.java:183)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.openpnp.vision.pipeline.CvPipeline.<clinit>(CvPipeline.java:49)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.openpnp.machine.reference.vision.ReferenceFiducialLocator.createDefaultPipeline(ReferenceFiducialLocator.java:353)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.openpnp.machine.reference.vision.ReferenceFiducialLocator.<init>(ReferenceFiducialLocator.java:50)
2018-10-01 20:01:57 SystemLogger ERROR: at
org.openpnp.machine.reference.ReferenceMachine.<init>(ReferenceMachine.java:89)
2018-10-01 20:01:57 SystemLogger ERROR: ... 46 more

vonnieda, 2018-10-02:
Looks like I might have built the new OpenCV with ffmpeg support by
accident. I'll probably need to push a new release of the library, but in
the meantime you can probably get past it with `brew install ffmpeg`.

Jason

wholder, 2018-10-02:
Thanks.  That seems to have worked, as the I can now get the GUI window to
display.  BTW, I went ahead and filed a bug report, anyway, just to help
inform anyone else.

Wayne

commit 32544455 (2018-10-02): Move translations into resources, where they belong. This fixes #773.
commit 7f301105 (2018-10-06): Merge branch 'develop' into feature/chmt36va

* develop:
  Move translations into resources, where they belong. This fixes #773.
  Fixing an issue where an already present head offset will lead to wrong offset calculation, fixes #769
  Improved Russian translations.
  Updated translations with those provided by Oleg and Javier. Added Spanish to language menu.
  Update changes for keep alive PR and set keepalive on by default to not surprise users.
  Added option connectionKeepAlive in AbstractReferenceDriver and AbstractReferenceDriverConfigurationWizard to keep connections alive on Machine di [... cut]
commit 665252ab (2019-05-19): Merge branch 'develop'

* develop: (63 commits)
  Note about reverting OpenCV upgrade.
  Revert "Updates required to move to opencv-3.4.2"
  serial fix
  Set ScriptEngine's FILENAME key to improve error messages.
  Escape angle brackets in exception messages.
  ReferenceDriverCommunications now allows you to specify the line ending type. Options are LF (default), CR, CRLF. No UI yet, change in machine.xml.
  Fixes #781 by changing the home icon indicator after the auto homing.
  Added welcome in Spanish
  Move translations into resources, where they belong. This fixes #773.
  Fixing an issue w [... cut]