openpnp/openpnp #774: Error while reading machine.xml

wholder, 2018-10-02:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
I'm trying to build OpenPnP using the latest version of IntelliJ IDEA in order to run and debug on my Mac Pro.

Info:
IntelliJ Community version 2018.2.4 (with Maven Plug in)
MacOS High Sierra 10.13.6
Java 1.8.0_144

## Steps to Reproduce

Import project into IntelliJ Community Edition using File>New->Project From VersionControl->Git
Build project.
Run project

## Expected Result
I expected the program to run.

## Actual Result
Code displays error dialog (I would add a screen shot, but that does not seem to be supported here.)  So, see the the stack trace below, instead.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
2018-10-01 20:01:14 Configuration INFO: No packages.xml found in configuration directory, loading defaults.
2018-10-01 20:01:26 Configuration INFO: No parts.xml found in configuration directory, loading defaults.
2018-10-01 20:01:35 Configuration INFO: No machine.xml found in configuration directory, loading defaults.
2018-10-01 20:01:57 SystemLogger ERROR: java.lang.Exception: Error while reading machine.xml (/private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib: dlopen(/private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib, 1): Library not loaded: /usr/local/opt/ffmpeg/lib/libavcodec.58.dylib
  Referenced from: /private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib
  Reason: image not found)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.model.Configuration.load(Configuration.java:287)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame.<init>(MainFrame.java:611)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.Main$1.run(Main.java:152)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch$$$capture(InvocationEvent.java:311)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(EventQueue.java:97)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:709)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:703)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2018-10-01 20:01:57 SystemLogger ERROR: Caused by: java.lang.reflect.InvocationTargetException
2018-10-01 20:01:57 SystemLogger ERROR: 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
2018-10-01 20:01:57 SystemLogger ERROR: 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
2018-10-01 20:01:57 SystemLogger ERROR: 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.InstanceFactory.getObject(InstanceFactory.java:100)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.InstanceFactory$ClassInstance.getInstance(InstanceFactory.java:233)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.ObjectInstance.getInstance(ObjectInstance.java:95)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.ObjectInstance.getInstance(ObjectInstance.java:76)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite$Builder.read(Composite.java:1376)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.read(Composite.java:201)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.read(Composite.java:148)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.readVariable(Composite.java:623)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.readInstance(Composite.java:573)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.readUnion(Composite.java:549)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.readElement(Composite.java:532)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.readElements(Composite.java:445)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.access$400(Composite.java:59)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite$Builder.read(Composite.java:1383)
[... 5 lines cut ...]
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Persister.read(Persister.java:584)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Persister.read(Persister.java:543)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Persister.read(Persister.java:521)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.simpleframework.xml.core.Persister.read(Persister.java:426)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.model.Configuration.loadMachine(Configuration.java:416)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.model.Configuration.load(Configuration.java:280)
2018-10-01 20:01:57 SystemLogger ERROR: 	... 17 more
2018-10-01 20:01:57 SystemLogger ERROR: Caused by: java.lang.UnsatisfiedLinkError: /private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib: dlopen(/private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib, 1): Library not loaded: /usr/local/opt/ffmpeg/lib/libavcodec.58.dylib
  Referenced from: /private/var/folders/mt/7rrh06nw8xj05b0052g526300000gn/T/opencv_openpnp1109948153633665499/nu/pattern/opencv/osx/x86_64/libopencv_java342.dylib
  Reason: image not found
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
2018-10-01 20:01:57 SystemLogger ERROR: 	at java.lang.System.loadLibrary(System.java:1122)
2018-10-01 20:01:57 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:208)
2018-10-01 20:01:57 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:189)
2018-10-01 20:01:57 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader$Holder.<clinit>(OpenCV.java:229)
2018-10-01 20:01:57 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader.getInstance(OpenCV.java:233)
2018-10-01 20:01:57 SystemLogger ERROR: 	at nu.pattern.OpenCV.loadShared(OpenCV.java:183)
[... cut ...] line.java:49)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.machine.reference.vision.ReferenceFiducialLocator.createDefaultPipeline(ReferenceFiducialLocator.java:353)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.machine.reference.vision.ReferenceFiducialLocator.<init>(ReferenceFiducialLocator.java:50)
2018-10-01 20:01:57 SystemLogger ERROR: 	at org.openpnp.machine.reference.ReferenceMachine.<init>(ReferenceMachine.java:89)
2018-10-01 20:01:57 SystemLogger ERROR: 	... 46 more

vonnieda, 2018-10-02:
I'll need to recompile opencv 3.4.2 without ffmpeg support, or maybe include the binary if I can find out how to do it legally. In the meantime, quick fix is `brew install ffmpeg`

vonnieda, 2018-11-17:
This was fixed in https://github.com/openpnp/opencv/issues/38

closed 2018-11-17 by vonnieda