openpnp/openpnp #776: Placement-error after Discard operation

ldpgh, 2018-10-03:
# Problem Report ... maybe feature request
Placement-error after Discard operation

## Describe the Issue
Discard location xyz=0/0/0  triggers my y-home-switch.
With nozzle-offset  xyz=-30.694/1.108/0.0 mm  and  discard-location xyz=0/0/0 mm the machine runs against the y-home switch for about 1.108 mm.

Due to the short failure distance it is hard recognize the root cause while debugging the pick location on the feeder and having vision enabled.

Possible workarounds:
- Discard-Location should take into account the nozzle-offset.
- sanity check to stay within the allowed region ... whatever this could mean (e.g. x>=0, y>=0)
- Change unassigned location from 0/0/0  to xyz=100/100/0. It should cover (hopefully) all nozzle offsets.

## Steps to Reproduce
1. Place camera above a location like the strip feeder and keep the coordinates (e.g. feeder camera location)
2. Issue a Discard operation
3. Move head again to the desired location using the same coordinates as used in 1.

## Expected Result
Location in 1. and 3. should be identical.

## Actual Result
Location of 3. is different by the nozzle offset.

vonnieda, 2018-10-03:
You can (and should) set the discard location to whatever you want: https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration%3A-Discard-Location

Does this not fix the problem?

ldpgh, 2018-10-03:
Change of Discard-location fixes the issue.

Close the issues ... will add a warning to the firmware if coordinates are below the home-coordinates.

closed 2018-10-03 by ldpgh